moonlight-stream/moonlight-embedded #158: Multiple controllers question

mercier1200, 2015-07-03:
Hi,
First I wanted to thanks everyone who works on this nice piece of software! I really enjoy steaming my games from my pc to the pi (I use raspbian).

I easily configured moonlight 2.0.1 with one of my wireless xbox 360 controller (with combo mapping/input) and it works in every games.

However, I cannot configure two wireless xbox 360 controllers when launching the stream.
I tried several forms and none of these works. Here is what I tried:
(this line…)
Moonlight steam -720 -60fps 192.168.8.119 -app “Trine”
( …combine with)
-input /dev/input/event5 -input /dev/input/event6
-mapping xbox.map -input /dev/input/event5 –input /dev/input/event6
-mapping xbox.map -input /dev/input/event5 -mapping xbox.map -input /dev/input/event6
-mapping xbox.map -input /dev/input/event5 -mapping xbox2.map -input /dev/input/event6

Xbox.map and xbox2.map are the same by the way. And I’m certain that both my controllers work because when I type:
-mapping xbox.map -input /dev/input/event5
Or
-mapping xbox.map -input /dev/input/event6

The right controller is detected by moonlight. Finally, I use the third method described in https://github.com/RetroPie/RetroPie-Setup/wiki/Setting-up-the-XBox360-controller to setup the controllers. Here is a cat of my last two /proc/bus/input/devices if needed:
I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="Xbox Gamepad (userspace driver)"
P: Phys=
S: Sysfs=/devices/virtual/input/input8
U: Uniq=
H: Handlers=js1 event5 
B: PROP=0
B: EV=b
B: KEY=7fdb0000 0 0 0 0 0 0 0 0 0
B: ABS=3001b

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="Xbox Gamepad (userspace driver) #2"
P: Phys=
S: Sysfs=/devices/virtual/input/input9
U: Uniq=
H: Handlers=js2 event6 
B: PROP=0
B: EV=b
B: KEY=7fdb0000 0 0 0 0 0 0 0 0 0
B: ABS=3001b

I guess my problem is simply a wrong usage of the parameters. I hope someone has tested this feature and can confirm the right spelling.
Thanks!

mercier1200, 2015-07-09:
I was asking this because I seen this commit ff344cfc30d692f210ba9e3b866f2f38579d8627 from @cgutman from the java implementation. My skills in coding are not particularly good so I'm not certain  to understand from the code how to use it. Reading the new C code I can understand this feature was ported to the new version.

Can somebody confirm this is working on his side?

cgutman, 2015-07-16:
It is working for me.

mercier1200, 2015-07-16:
Really?!
1) How exactly did you call the mapping/input combo on the command line?
2) Are you using wireless Xbox 360 controller as well? If not, which ones?
3) Now that I think, Trine was the only game I tried. Maybe there is a problem with this one. Which game you tried to test this feature?

mercier1200, 2015-07-16:
Other question ;)
4) Which argument are you using with xboxdrv (assuming you are using it)? Do you have a special argument not listed into the link I've posted in my original post?

7hunderbug, 2015-07-16:
It works for me too with one Wireless Xbox 360 controller and one Nintendo Wiimote (connected via bluetooth).  I've played numerous two player games like this (Ibb & Obb, Guacamele to name just two).

Your moonlight command line doesn't need ```-720``` or ```-60fps``` as these options are the default anyway.

Run moonlight something like this for multiple controllers:
```moonlight stream -mapping xbox1.map -input /dev/input/event0 -mapping xbox2.map -input /dev/input/event1 PC-IP-address-or-hostname```

Also, I would try moonlight without forcing a specific game with```-app``` first, so that Steam runs and detects your two controllers. You can then choose which game you want to play in Steam Big Picture Mode.

mercier1200, 2015-07-17:
Thanks @7hunderbug !

It is indeed working directly through stream, I don’t know why Trine (as a non-steam game) is not working but Trine 2 (as a steam game) works flawlessly. Even with the –app “Trine 2” the game recognized the 2 controllers.

My problem was more game-oriented than moonlight-oriented.

Thanks again for your time

closed 2015-07-17 by mercier1200

Haris1977, 2015-10-23:
moonlight stream -mapping xbox.map -input /dev/input/event0 -mapping logitech.map -input /dev/input/event1 PC-IP-address-or-hostname doesnt work for me (system hangs in Can't open configuration file hosts)

I do not have 2 xbox controllers though. I use a wired logitech rumblepad 2 and a wireless xbox compatible controller.

Any fixes?

mercier1200, 2015-10-23:
@Haris1977 This was fixed by irtimmer a month ago in #215.

Just wait for the next release or compile the source yourself and you should be in business.

Haris1977, 2015-10-23:
That's good news! Thanks!

Haris1977, 2015-10-29:
Any estimated date for the new release?? 

Can't wait for the controller fix! I also have too many errors in compiling it from beginning so we keep our fingers crossed for the new release..