moonlight-stream/moonlight-embedded #160: "poor network connection"

Keda2, 2015-07-05:
huhu and sorry for my bad language :>

i had different problems with limelight in the past days  and don't know what i can try next.
first i had problems with the rtsp error (400). i reinstalled the nvidia drivers and thought i'm done know, but then there was the next error. 

I'm using GFE 2.4.5.57 with a GTX960 and limelight 1.2.2. My computer and the Pi2 are wired connected (1GBit / 100MBit). 
i tried the moonlight app for my samsung galaxy s4 and it workes fine, without delays. but when i start it on the pi, it always shows "poor network connection" or sometimes connects. i can see and navigate Steam then, but after a few seconds the screen freezes for some seconds. i tried to change the /boot/config.txt from the pi, but it was empty, so i added "gpu_mem=256" to it. in the raspi-config there always shows up an error when im trying to change the cpu/gpu options...

I also tried to set the bitrate to 25000 and update/upgrade the pi and java, but nothing happened.

This shows when im starting the stream:

 $ java -jar limelight.jar -720 -30fps -bitrate 25000 -app Steam stream 192.168.178.xx
Starting Steam
Starting RTSP handshake
Starting control connection
Starting video stream
Starting audio stream
Starting input connection
Poor network connection
Poor network connection
java.net.SocketException: Broken pipe
        at java.net.SocketOutputStream.socketWrite0(Native Method)
        at java.net.SocketOutputStream.socketWrite(SocketOutputStream.java:109)
        at java.net.SocketOutputStream.write(SocketOutputStream.java:153)
        at com.limelight.nvstream.control.ControlStream$NvCtlPacket.write(Unknow                               n Source)
        at com.limelight.nvstream.control.ControlStream.sendPacket(Unknown Sourc                               e)
        at com.limelight.nvstream.control.ControlStream.sendLossStats(Unknown So                               urce)
        at com.limelight.nvstream.control.ControlStream.access$100(Unknown Sourc                               e)
        at com.limelight.nvstream.control.ControlStream$1.run(Unknown Source)
java.net.SocketException: Socket closed
        at java.net.PlainDatagramSocketImpl.receive0(Native Method)
        at java.net.AbstractPlainDatagramSocketImpl.receive(AbstractPlainDatagra                               mSocketImpl.java:143)
        at java.net.DatagramSocket.receive(DatagramSocket.java:781)
        at com.limelight.nvstream.av.video.VideoStream$1.run(Unknown Source)
java.net.SocketException: Socket closed
        at java.net.PlainDatagramSocketImpl.receive0(Native Method)
        at java.net.AbstractPlainDatagramSocketImpl.receive(AbstractPlainDatagra                               mSocketImpl.java:143)
        at java.net.DatagramSocket.receive(DatagramSocket.java:781)
        at com.limelight.nvstream.av.audio.AudioStream$1.run(Unknown Source)
Connection Terminated The connection failed unexpectedly

sometimes it only shows "poor network connection" again and again until i close steam on my pc. my smartphone is wireless connected with 65MBit, why does it work without problems, it shouldn't be my router then, right? My upload is bad, but when it's only in my home's network, i think it shouldn't be the problem?

I really have no idea and hope you can help me :>

schulzad, 2015-07-06:
How is you Raspberry Pi connected to your network? Wireless or Wired?

Are you running Raspberry Pi B+?

 I've also had to overclock it to get decent performance. You can overclock it via "rasp-config"

cschlonsok, 2015-07-09:
I have exactly the same problem like Keda2. 
First tried it wired with Pi2 (100/1000 MBit) directly connected to the router. With 720p at 30 fps. Was completely unplayable. 

Now I tried it with a USB wifi stick with up to 300 Mbit 5G. And now it is way more playable. Every 2-3s it just lag a bit.. This playing is still way worse than every youtube video i have seen so far. 

The game is running fine on the computer. I can play it with the controller connected to the Pi2.

The file transfer from my Windows Pc to my Pi2 (Samba Sharing) is about 10 MB/s that's fine for me. But sometimes it drops for some seconds to 200 kb/s or less. 
Maybe the Pi itself has a program running which is making these Problems. I don't know what else is making these problems. Does anybody have an idea how I could check that the best way?

irtimmer, 2015-07-09:
GFE >= 2.4.5.54 isn't supported by the old (and no longer supported) Java based Moonlight Embedded. Please update to a newer version.

closed 2015-07-09 by irtimmer