moonlight-stream/moonlight-embedded #161: Overscan

bentglasstube, 2015-07-05:
When I start streaming, moonlight seems to ignore my overscan settings in `/boot/config.txt` so many games are made difficult by the fact that hud elements are displayed.  I looked through the source but didn't see anything to set up overscan.

irtimmer, 2015-07-07:
Video is often displayed without overscan (the reason why most logo's and subtitles are not placed to the edge). Therefore video is also not configured by config.txt and must be programmed seperetaly in the application. Which isn't (yet) the case in Moonlight Embedded.

BrainSlugs83, 2015-11-24:
+1 -- this is pretty annoying in most games, and it even makes some games unplayable (because you can't read the text, etc.).

Haris1977, 2015-11-24:
+1 for this also. It is pretty annoying

richterh, 2015-11-29:
I found a solution that corrects the overscan on the OS level rather than in moonlight itself. This was tested on a Raspberry Pi B+ running OSMC.

As @irtimmer stated, the video is a separate component, but there is in fact a setting in config.txt that applies the overscan correction to all video output. So, to apply the config.txt correction to all video output, add the following line to /boot/config.txt:

`output_scale=1`

(You also need to set `disable_overscan = 0` and the actual correction values, see below)

To the best of my knowledge overscan settings will now be applied to all video output.

For more information see:
https://www.raspberrypi.org/forums/viewtopic.php?t=35971&p=306184
https://www.raspberrypi.org/forums/viewtopic.php?f=35&t=25993

An example of the relevant parts of my config.txt:

```
overscan_scale=1
disable_overscan=0
overscan_left= -9
overscan_right= -9
overscan_top= -28
overscan_bottom= -28
```

This also corrects the overscaling in SMC/KODI, so it doesn't need to be set using *Settings -> System -> Video Output -> Video Calibration* anymore.

closed 2016-04-03 by irtimmer