moonlight-stream/moonlight-embedded #162: Streaming on my self PC and not TV

p2pow350, 2015-07-06:
Hi,
I've succesfully compiled moonlight-embedded under a Raspberry PI 1.

I can pair my raspberry with the PC on same LAN, I can see a list of games supported, however when I run the command to open the stream the screen start on my self pc and not on the remote raspberry.
Am I missing something?

The command I'm running is  "moonlight stream -audio 1 -app Steam -1080"

In this example, Steam start but on the PC and not on the PI :(

Thanks
Max

PietroDR, 2015-07-07:
Hi, 

I have the same issue on a raspberry pi 2 : I can't see anything on the distant screen but I'm able to ear the sound of the Steam app. 
I'm on OSMC, I've compiled from the latest release with no errors. 

Thanks if someone has the solution.

irtimmer, 2015-07-07:
You probably haven't installed the Raspberry Pi library headers. Check when running cmake if BROADCOM libraries are found. For OSMC see also my reaction in #150

p2pow350, 2015-07-07:
Hi,
In my opinion I've installed all libraries needed including user land (found on github), but maybe I'm wrong...
Please check if when compiling is missing something (seems all ok):

root@xbian ~/src/moonlight-embedded/build # cmake ../;make;make install
-- The C compiler identification is GNU 4.6.3
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CURL: /usr/lib/arm-linux-gnueabihf/libcurl.so (found version "7.26.0") 
-- Found OpenSSL: /usr/lib/arm-linux-gnueabihf/libssl.so;/usr/lib/arm-linux-gnueabihf/libcrypto.so (found version "1.0.1e") 
-- Found EXPAT: /usr/lib/arm-linux-gnueabihf/libexpat.so (found version "2.1.0") 
-- Found ALSA: /usr/lib/arm-linux-gnueabihf/libasound.so (found version "1.0.25") 
-- Found Opus: /usr/local/lib/libopus.so  
-- Found Broadcom: /opt/vc/include  
-- Could NOT find Freescale (missing:  FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY) 
-- checking for module 'libcec>=3.0.0'
--   package 'libcec>=3.0.0' not found
-- Could NOT find CEC (missing:  CEC_INCLUDE_DIRS CEC_LIBRARIES) 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'libevdev'
--   found libevdev, version 1.4.3
-- checking for module 'avahi-client'
--   found avahi-client, version 0.6.31
-- checking for module 'libudev'
--   found libudev, version 175
-- Configuring done
-- Generating done
-- Build files have been written to: /root/src/moonlight-embedded/build

It's strange that the stream start on the PC, I can see the Nvidia Experience starting, but it's opening on the PC and not on the PI. This running the stream command with SSH or directly on the TV with a USB keyboard..

Thanks for you great project!
I really would like to open games on my TV and not on my "small" screen PC.

irtimmer, 2015-07-07:
In that case their should at least some video output or a error message, otherwise I have no idea what's going on.

PietroDR, 2015-07-07:
Ok, so we have different problems. In my case, installation of raspberry pi library headers solved the problem (`sudo apt-get install rbp-userland-dev-osmc` for OSMC). 
Thank you very much !

p2pow350, 2015-07-07:
Is there a way to enable a log or does moonlight itself write a log file?

7hunderbug, 2015-07-16:
@p2pow350 do you have a secondary audio device connected to the Pi? If not then you shouldn't need to specify the "-audio" option. Have you allocated enough memory to the Pi's GPU? Also, moonlight/NV streaming runs Steam by default so you don't need the "-app" switch either. 1080p streaming doesn't work unless you over-clock your Pi so I would not use the "-1080" switch either. I would try running moonlight in as simple a configuration as possible, like this: ```moonlight stream PC-IP-address-or-hostname```

closed 2016-01-16 by irtimmer