moonlight-stream/moonlight-embedded #163: Ignoring terminal input

vulcandth, 2015-07-12:
After exiting moonlight from playing games on my raspberry pi, i'm left with lines upon lines of input left in my shell window filled with characters typed on my keyboard while playing my game. Example (aaaawwwwwwssswwwaaaaawwdddaaaasssss) 

If its possible we should find a way to block input into the background shell.

Let me know if I can explain this any better.

Lombra, 2015-07-13:
I have this too. What's even worse is that when I navigate some menu (eg Steam) using the keyboard, these keystrokes will be queued up to be executed once Moonlight exits, going through recently entered console commands (up arrow) and executing them!! (enter) That's potentially very harmful.

vulcandth, 2015-07-14:
Sorry accidentally hit the close button lol.

darkpavian, 2015-07-14:
yep, that's right. E.g. when using the keyboard to shutdown the stream it is Arrow+up, [Arrow+right, Arrow+right], return, which causes the moonlight command to be executed imidiatly again after shutdown.

vulcandth, 2015-07-14:
I have a couple idea of how to fix this; When I get home from work i'll fork this repo add my changes, test, and suggest a merge.

vulcandth, 2015-07-15:
#166

irtimmer, 2015-07-15:
Should now be solved by #167

closed 2015-07-15 by irtimmer