moonlight-stream/moonlight-embedded #164: Bad stream quality

darkpavian, 2015-07-13:
Hi everyone, 

I don't know if this is a directly moonlight related issue, but since I cannot localize the problem itself, I just wanna try to get some input here and maybe share some information to people with similiar issues. 

My Setup:
PC: i5-750, 8gb RAM, GTX 770
Client: RPI 2, Raspbian (Kernel 4.0), using Gigabit-USB-Ethernet Adapter, xbox 360 wireless and small rapoo wireless keyboard
Network: Fully wired. 

iperf put out about 130 to 160 mbit using the Gbit usb adapter, and almost 100 mbit using the build in ethernet. 
From what i have read this should be enough bandwith for at least a -720 -30fps stream, but any configuration -720/-1080 and -30fps/-60fps just gives me a really but image quality when streaming. 

Is there any configuration I am missing? Maybe any configs to be done on my PC?
May this be an issue related with the usb/ethernet controller of the raspberry pi? (Afaik there is only one controller for usb and ethernet, so that all the bandwidth (~480mbit) is split between all devices connected.)
May this be a network problem? 
Or may this issue be game related? (GTA5 ran a little better than Witcher3 which is almost unplayable)

Best regards and many thanks for any help provided, 
Florian

HazCod, 2015-07-13:
Try looking at cpu/mem/network consumption on both the pi and pc when streaming

darkpavian, 2015-07-13:
@HazCod Thanks for the handy tools. I did not know them before. 
After some tests i got the following results:

moonlight @ 720p / 60fps
[image]

limelight @ 1080p / 30fps
[image]

The consumption looks pretty simliar ofr any configuration / tool I use.

darkpavian, 2015-07-14:
First I thought that the dot in the values is the thousands seperator, because I have set the locales to german (that is how it is used in germany) but it seems it is locale-independent. So the question is, why is moonlight only using this little bit of available bandwidth?

darkpavian, 2015-07-15:
Update:

This is the bandwith iperf uses. 
[image]

I also tried setting up a clean system on another sd card, building moonlight from source, but without success.

wkewlin, 2015-07-15:
Seems like you have a network problem.
On Pi 2 and wired network you should see 100Mbit saturated by iperf as in my screen shot
[image]

darkpavian, 2015-07-15:
Thanks for the idea. 
I checked the speed with iperf in the other direction. I have got the server running on my win pc and using the pi as the client. So the higher value is the sending bandwidth with my setup. 
Testing it a few times show the same values as you get them, just swaped for SENT/RECEIVED.

wkewlin, 2015-07-15:
ok just realized myself that you checked other way around.
But are you using the old java client? 
Did you try the new c client?
It is still strange, because the old one worked also fine for me (at least for 720p/60fps)
Other thing to consider, gamestream relies for some reason on the resolution of windows monitor as in you cant go higher than set there. The workaround for that is to define custom resolutions. But 720 should have been still fine...

wkewlin, 2015-07-15:
[image]
[image]

Just checked, in Wither 3 menu i have even less network and cpu usage, so this seems normal

darkpavian, 2015-07-15:
Thanks for all the input.

I am currently running the latest version of moonlight, freshly compiled on my pi. I also tried the old java client. 

You might be right about the desktop resolution. My PC is running headless downstairs and I used xtightvncviewer to view the desktop. This might have changed the resolution. I just checked it and it was set to 1024x768. So I set it back to 1080p. 

Now the network starts dropping lots of Frames. I get many messages like:

Network dropped an entire frame
Rsync complete

darkpavian, 2015-07-15:
Update: 
Due too all the dropped Frames, the steam big picture menu is laggy as hell. Funny thing: As soon as the game started, it runs smoothly, but although i changed the desktop resolution, the game stream quality does not look better.

wkewlin, 2015-07-15:
From my experience gamestream really does not like headless. For some reason it checks the max resolution of attached monitor and does not allow to go higher. To circumvent this you have to define custom resolution in nvidia settings. I had to do this to stream in 1080ß with an 1680x1050 monitor attached, without custom resolution I could stream in 1080p but games would run in 720 or other settings.
And you need to overclock the gpu of the pi for 1080p to work or it is laggy as hell (like 2sec lag)

darkpavian, 2015-07-15:
Ok, this is interesting because I played GTA 5 completely with the headless setup, without any problems. Witcher 2 also worked fine. 
I cannot totally determine the time when the problems start, but it may have started when I switched from using teamviewer to using VNC. 
The desktop resolution is still set to 1920x1080, as my monitor uses this as native, so I don't think I have to create a custom resolution. 
I will wire my monitor back to my PC tomorrow to check if this will reconfigure anything.

wkewlin, 2015-07-15:
btw nethogs is useless for this test. just watched a youtube chanel through moonlight (1080/60 at 20MBit/s) and running taskmanager in windows I can see that it sends about 1MB/s of data and nethogs never picks it up.

ZC2K, 2015-07-16:
I have similar problems with my PI2 like the darkpavian ...

My Setup
PC -> Xeon E3-1231v3, 8GB Ram, GTX 970
PI2 -> USB wireless Logitech Keyboard, USB wireless XBox360 Controler, USB WLAN 300 MBit Adapter

I am using OSCM and compiled a new Build from the latest Moonlight Sources. With iperf i have around 40MB/s and streaming from my NAS works like a charm. 
However, if i set moonlight bitrate to 2000 then everything runs smooth without lag but looks bad. If i raise the bitrate or use the default (8000) then it is laggy as hell.
Could it be the netcode, the streaming code, the Pi2 performance ? It does not matter if it is connected via WiFi or Ethernet.

darkpavian, 2015-07-16:
Are you using your PC headless or do you have a monitor connected to id?

I couldn't try it, yet, but from what I have read wkewlin might be right. Some people are encountering similar problems streaming via Steam In-Home streaming from headless servers. 

There is a little piece of hardware that was developed because of this problem: https://www.headlessghost.com/

ZC2K, 2015-07-16:
@darkpavian: I have a 2560x1080 Monitor connected to my PC (DisplayPort) which is also running this resolution at the desktop and in every game. I have the Pi2 also connected to this Monitor via HDMI for testing (Picture in Picture). The Pi2 is also around 1 meter away from the wireless Router (AVM FritzBox 7360)

darkpavian, 2015-07-16:
@ZC2K Sounds familiar. I also tried setting the bitrate manually. In addition to the lagging, I get crashes due to segmentation faults pretty often.

darkpavian, 2015-07-16:
I have just tried to re-head my pc and run moonlight, but it looks nothing better then before.

slawak, 2015-07-16:
I have also an AVM router (7490). They did mess up things few times in the past, but usually with WLAN. It would be best to do the test with Pi and PC connected directly to a switch to really exclude any network problems.

darkpavian, 2015-07-16:
I am only using wired networks. As I stated before, it was running fine all the time and suddenly from one day to another the image quality completely broke down. 
I did not change anything about the connections between Pi and PC. 

Unfortunately I am not able to wire the pc directly to the router, because it is in the basement, the router on the second floor.

slawak, 2015-07-16:
You should also assign static IP addresses to PI and PC and call moonlight with IP and not with hostname. The DHCP server and DNS server in AVM boxes do have some strange quirks. (My PI gets different addresses went booted with different sdcards and I can see those in the ARP cache of my PC, resulting in bad behaviour.)

darkpavian, 2015-07-16:
Yes, you are completly right. I already assigned most of our networking devices (ATV, AVR, TV, FireTV, etc.) static IP Devices on client side, and checked for all the devices in the FritzBox settings to always assign them the same IP adress. 

I don't think that it is a Pi only problem, as the lagging due to frames dropped by network also happens on my iPad.

7hunderbug, 2015-07-16:
Could you post your Raspberry Pi's /boot/config.txt please so we can check for issues?

I would try the following to eliminate them from being possible causes (similar steps helped me with bad stream quality before).

1. Remove your Gigabit-USB adapter (not required at all for moonlight streaming) and use the Raspberry Pi's built in 100Mb ethernet port.
2. Remove your wireless keyboard receiver and run commands on the Pi using putty or SSH from another device.
3. Reinstall *ALL* Nvidia software on your PC, including GFE, GFX card drivers and audio drivers. When running the Nvidia installer, make sure you select "Custom" install so that you can check the option to perform a "Clean Installation".

slawak, 2015-07-16:
For what its worth I just updated to latest AVM firmware (6.30) and it still working fine. But I had once a defective AVM box which went to 100 Mbits and sometimes 10 Mbits on a 1 Gbits wired network and occasionally dropped the connection altogether. Had to sent it in as it was a hardware issue.

As you have issues with iOS client as well, I still suspect its some strange network problem or maybe its the PC. So reinstalling Nvidia software as 7hunderbug suggests is a good idea.

7hunderbug, 2015-07-16:
In addition, I would also try the following to eliminate them from being an influence to your problem:

1. Ensure that your PC desktop resolution is either 1280x720 or 1920x1080.
2. Ensure that the resolution settings in Steam Big Picture mode (Under Display->Active resolution) matches the resolution of your desktop (so either 1280x720 or 1920x1080).
3. Uninstall all remote desktop software from your PC. (Remove VNC if you use VNC and/or Teamviewer if you use that).

darkpavian, 2015-07-16:
Many things to try. Thanks a lot for the ideas. 

This is the /boot/config.txt:
# For more options and information see
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
arm_freq=1000

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

core_freq=500
sdram_freq=500
over_voltage=2
gpu_mem=512

I removed all unnecessary USB devices and reinstalled any Nvidia software, that at least eliminated the frame dropping issue. 
My Desktop resolution is 1080p same as the resolution setting in Steam Big Picture Mode. 
I uninstalled the VNC server, but I can not uninstall teamviewer. It is my only connection to the pc, which is still running headless in the basement. I will try that tomorrow.

7hunderbug, 2015-07-16:
You shouldn't need any over-clocking options in your /boot/config.txt unless you are trying to get 1080p streaming working. In my opinion 720p streaming is more than adequate and is more reliable in terms of quality. Therefore the only option you require in your config.txt is ```gpu_mem=512```.
Either remove or comment out the following:
```
arm_freq=1000
core_freq=500
sdram_freq=500
over_voltage=2
```
My gpu_mem is set to 256 and streaming works fine. Perhaps your should try changing yours from 512 to 256 to see if that helps?
Also, I would try moonlight with default settings, so don't specify -720, -1080, -30fps, -60fps, -bitrate just run it like this: ```moonlight stream PC-IP-Address-or-hostname```

7hunderbug, 2015-07-16:
One last thing to try: Turn off all wired and wireless devices that are not essential to moonlight streaming, ones that would normally be connected to your router, to ensure that they are not interfering with traffic through your router.

slawak, 2015-07-16:
Ok I just updated GFE to 2.5.11.45 and this lead immediately to choppy video. Not so much lag but only few frames per second on Pi, everything normal on PC. 
Deinstalling GFE and installing 2.4.5.47 solves the problem.
The change notes of GFE mention "Incremental improvements in GameStream audio latency." and the menu is called SHIELD and not Gamestream anymore. So maybe there were bigger changes.
Pls check your GFE version and try a downgrade.

scoobyd00, 2015-07-16:
Can anyone tell me where i can download an older version of geforce experience (2.4.5.47).

The latest version i have auto updates and theres no way of stopping it, unless someone knows how?.

7hunderbug, 2015-07-16:
Download an older driver package from Nvidia - the package comes with 
everything in it. Then run a newer Nvidia installer in "Custom Install" 
mode - you can then selectively upgrade the drivers but leave GFE at the 
older version.

scoobyd00, 2015-07-16:
Cheers, do you know why driver version contains the best geforce experience so it doesnt autoupdate?.

7hunderbug, 2015-07-16:
I'm not sure, you'll have to read the driver information on the Nvidia 
page to determine which version of GFE is shipped with it. GFE 
autoupdates if it is set to do so - I think you can disable it.

I use the most recent drivers and GFE in my setup with a Geforce GTX 670 
and it all works flawlessly so I doubt downgrading GFE will fix your 
problem. Certainly, I've tried downgrading in the past thinking this 
would solve my streaming issues but it only made minor improvements. 
What did help is a completely clean install of the most recent NV 
drivers and GFE - upgrading via the upgrade method provided by either 
the GFE application or the NVidia software nearly always resulted in bad 
streaming performance for my setup, which is why I have resorted to 
always doing a clean install when NVidia release new drivers.

scoobyd00, 2015-07-16:
Thanks for the update, my setup is working great but ive just read slawak post about the very latest experience causing issues for him. I havent had chance to test it yet but the later versions of experience automatically update and theres no way to stop it. As soon as you click on the geforce experience it runs the update and if you say decline then you cant get into the program.
Im downloading some older drivers now just incase in the future there are issues with the latest versions of experience.

7hunderbug, 2015-07-16:
Ah, oops, I didn't realise a new update has just been released for GFE! It is possible the new one has problems. I'm on 2.4.5.57. You can prevent GFE from auto-updating by running it directly from C:\Program Files (x86)\NVIDIA Corporation\NVIDIA GeForce Experience\GFExperience.exe but you're right, you can't disable it (I was thinking of the driver updates which you can delay).

7hunderbug, 2015-07-16:
Nvidia drivers 353.30:
http://uk.download.nvidia.com/Windows/353.30/353.30-win8-win7-winvista-desktop-release-notes.pdf
~page 9 "Software Modules" - "GeForce Experience - version 2.4.5.28"

Previous GFE available here: 
http://uk.download.nvidia.com/GFE/GFEClient/2.4.5.57/GeForce_Experience_v2.4.5.57.exe

scoobyd00, 2015-07-16:
Driver version 347.88 contains Geforce Experience 2.2.2 which is the last version of experience which doesnt include the auto update for information incase the latest experience version doesnt work.

7hunderbug, 2015-07-16:
Have just upgraded to GFE 2.5.11.45 and performance is virtually identical to previous version. I maybe get one or two audio dropouts occasionally, perhaps every couple of minutes but nothing major.

scoobyd00, 2015-07-16:
I've just experienced the same, the odd audio drop out.
just tested geforce experience 2.2.2 and driver 353.30 and it runs smoothly, takes abit longer for the stream to start but no issues when running.
think ill stick with this combination as it makes sure thr geforce experience doesnt auto update which may cause issues in the future

darkpavian, 2015-07-17:
Nice. I will try this out as soon as I am back home. 
I redirected the output of moonlight into a log file to check about any interesting messages
```
moonlight [...] > /home/pi/moon.log
```
and got some ALSA errors. This could match with nvidia changing something about the audio stream. 
Although the audio still worked fine for me. Even with stuttering video stream, the audio stream runs smoothly...

darkpavian, 2015-07-17:
I am running GFE 2.2.2 with Driver 353.30, same os scoobyd00, but without any luck. 
The Alsa errors I got are: 

```
Alsa shortm write, write 20 frames
Alsa shortm write, write 28 frames
Alsa error from writei: -11
Alsa shortm write, write 212 frames
Alsa error from writei: -11
Alsa error from writei: -11
Alsa shortm write, write 232 frames
Alsa error from writei: -11
Alsa error from writei: -11
Alsa shortm write, write 8 frames
Alsa error from writei: -11
Alsa shortm write, write 16 frames
Alsa error from writei: -11
Alsa shortm write, write 8 frames
Alsa shortm write, write 232 frames
Alsa error from writei: -11
```
And when trying to run with -1080 it crashed with
```
Can't empty video buffer
```
I will take the pi downstairs to the PC and test a direct connection now.

scoobyd00, 2015-07-17:
I get some of those alsa shortm write errors but it doesnt seem to effect gameplay or the sound streaming, not sure what they mean?.

Im running wireless from my pc upstairs to tv in living room. Im setting bitrate to 15000 and get good quality and streaming, 1080p is too much for my wireless connection though and i get serious lag still on a rpi 2. Ive got a good 5ghz connection to the router, full strength and can download at full speed on my 152mb broadband connection at 16mb/s, think 1080 is setting bitrate at 20000 which is too much for me.

darkpavian, 2015-07-17:
I am currently running out of options.

Pi related:
* It does not seem to be a PI/moonlight-embedded related problem. I compiled and ran in from a clean installation on the PI. Issue remains. 
* The streaming quality looks on the iPad as bad as on the the Pi.

Network related:
* It is not a network problem. I wired the PI dirctly to the second ethernet adapter of my PC (different subnet). Still no quality improvements. 

PC related:
* I first clean reinstalled all Nvidia PC software to current version, then downgraded to GFE 2.2.2 (everything done clean). 
* Tried removing everything unnecessery from the party, i.e. usb devices and overclocking stuff. 
 
Is it possibly a hardware issue of my gtx 770? The game runs without any problem with pretty high settings on the PC so I do not really assume any grafics card hardware problems...

7hunderbug, 2015-07-17:
Did you try removing Teamviewer? As far as I know, teamviewer (and some other remote desktop software) installs a graphics mirror driver that may interfere with gamestream operation. Other than that, because the bad experience is the same on your iPad and Pi, the problem must be with your PC or the software installed on it. What other applications run in the background and in the system tray on your PC?

ZC2K, 2015-07-17:
So ... the last 2 days i checked nearly every option mentioned here. I still have a lag no matter what (except i set -bitrate 2000). Wired connection is maxed out and my network is fine with all its components. I guess i have to wait for some new builds, WIndows IOT or Steamlink. Adventuregames and streaming HD content via Plex works fine though. Maybe it is OSMC itself (kernel,driver) ?

slawak, 2015-07-18:
Ok so GFE just self updated from 2.4.5.47 to 2.5.11.45 today again and I checked Witcher 3 again and again it has choppy video, but no audio problem as far as I can see.
The log is full of 
```
Network dropped an entire frame
Resync complete
Network dropped an entire frame
Resync complete
```
which it was not with 2.4.5.47. (Played Witcher 3 in 1080/60 for some time)
So at least for me there is a repeatable change for worse from 2.4.5.47 to 2.5.11.45.
Unfortunately I cant go back to the non auto update version 2.2.2 since new versions contain fixes for Witcher 3. It seems I have deinstall GFE 2.5.11.45 and install GFE 2.4.5.47 again and never give it the permission to update.
For reference I have Windows7 and Geforce TITAN (original version). Driver version was 353.30.

cgutman, 2015-07-19:
I tried GFE 2.5.11.45 today and didn't have any issues. I'm running a Pi 2 on the 4.0 kernel. Pi connected to the PC via Ethernet. Running h264_freq=333, force_turbo=1. 1080p60 streaming is working smoothly.

ZC2K, 2015-07-20:
@cgutman : What is your OS on the PI2 (which distribution do you use) and what does your network configuration look like ?
I have an AVM FritzBox 7390 on the ground-floor, an AVM Repeater (Bridge Mode) on the first-floor, and the AVM FritzBox 7360 in the atttic. Everything connected via Cat7 through a TP-Link Switch.
I think it is OSMC, because i can not understand why this lag happens via WiFi and Ethernet. I also get the "frame drop" message and Alsa errors from time to time.

cgutman, 2015-07-20:
@ZC2K I'm running Raspbian Wheezy with the 4.0 kernel. My network configuration is much more simple. I've got my Pi 2 connected directly to my PC via a long Cat6 cable. I don't see any frame drop messages with this configuration.

The Alsa errors are sometimes normal. The GFE will stop sending audio packets if there's no audio being played from the PC. This can cause underruns that lead to those messages in the logs.

darkpavian, 2015-07-21:
@7hunderbug I was not able to test anything during the last days, but will remove teamviewer tomorrow and try again. 
@cgutman Do you (or any body else) have any experience regarding any Remote Desktop tools in combination with moonlight?

darkpavian, 2015-07-22:
I literally striped down my system to the bare nudes. Removed anything that is not necessary for the stream, but did not gain any quality. I will resetup my windows system from scratch now, to try again.

ZC2K, 2015-07-24:
I did a clean install with Raspbian and Linux Kernel 4. Moonlight via apt-get but the issues are still the same. If i go beyond a bitrate of 2500, i get lags and packet/frame loss. No mather if Ethernet or WiFi. Tests with iPerf are still lokking good ... 1MB/s on Ethernet and 700kb/s on WiFi.

darkpavian, 2015-07-24:
Same here, I did clean installs with Raspbian, everything up-to-date, compiled moonligt from git repo, and also a clean installation of the PC. The one thing I did not reinstall is steam. Could this be a steam related issue? I checked the resolution of the big picture mode several times, but it seems to be ok...

ZC2K, 2015-07-24:
Hello Again. This time fresh install on PC with Windows 10 (10240) and still no change. I think it it not possible to get a decent stream above 720p/30fps/2500 bitrate.
The Info says it is kbps so if i go by the source code and the standard for 1080p/60fps is 20000 kbps i would need to transfer ~2,5MB/s for the stream only.
WiFi is someware between 2MB/s - 3MB/s
ETH is around 112 MB/s (!)
Both checked with iPerf (10x and took middle value).

irtimmer, 2015-08-15:
Newly released version has some fixes to minimize problems on higher bitrates. Please check if those fixes solve your issue.

spacewave, 2015-08-15:
I had slight graphical issues with streaming a while back, i found out that it was my Switch between my router and my pi.  

Once i removed the switch, everything was fine.

If someone could recommend a low latency switch, i'd really appreciate it.

darkpavian, 2015-08-15:
@irtimmer Dank je well for the info. I compiled the latest version, but could not see any major improvements. I tried different settings, starting from default. I will now setup the pi from scratch trying your prebuild package. 

@spacewave Thank you for the idea, but I already tried wiring the pi directly to a secondary networks adapter on my PC. So without any further switches/routers and stuff in between. Without any improvements.

cschlonsok, 2015-09-18:
@darkpavian Hi I had problems too. My stream freezed every ~2s for some ms. Made some games unplayable. My Pi2 was directly wired to my AVM Router and my PC too. 
I tried everything, but nothing really worked. This morning i checked the Ethernet-card settings on my Windows 8 PC and found some nice settings: There is a setting "Bandwidth & Duplex" which was set to "Automatic". After changing it to "100 Mbit/s Fullduplex" everything worked fine. Even 1080p 30fps were realy playable. I also found a setting called "Green-Ethernet" you may want to set this to "Off" too, cause we need some good performance ;)

The only problem i am now having is a sound delay of 2s but i can live with that :D

Hope it will help you too.

[image]

HazCod, 2015-09-18:
Could you do a wireshark capture on your computer? That really gives us more insight in what is happening.

darkpavian, 2015-11-08:
Hello again, because it looked quiet hopeless to solve this issue, I chose to focus on using the steam in-home streaming as my steam link will arrive shortly. 
During some testing in the last few days, however, I encountered excatly the same problem. Low quality, picture looked "bricky". 
As stated in this thread in the [Steam community] (https://steamcommunity.com/groups/homestream/discussions/0/483367798501608669/) forums, the problem might be NVidia related. 
When the hardware encoding on the host is switched on, the bandwidth is limited to 30 Mbits, no matter how good the network is. As soon as I switch to software encoding the bandwidth which is estimated by steam is somewhere around 700 - 800 Mbits in our Gbit network. 

Is there a way to force software encoding using the NVidia game stream?

[2 more replies]

closed 2016-01-16 by irtimmer