moonlight-stream/moonlight-embedded #165: Moonlight hard exit with 'Can't empty video buffer'

maverickjesus, 2015-07-14:
Had this a few days ago playing Renegade Ops (but didn't have chance to take a copy of the output), and triggered it again today playing The Witcher 3 for about 20 minutes. The moonlight process hard exits and dumps back to the console.

pi@raspberrypi ~ $ sudo moonlight -1080 -30fps -mapping xbox.map -input /dev/input/event0 -app Steam stream
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Discarding RTP packet queued for too long
Network dropped beginning of a frame
Waiting for IDR frame
Resync complete
Alsa shortm write, write 16 frames
Alsa error from writei: -11
Alsa error from writei: -11
Alsa error from writei: -11
Discarding RTP packet queued for too long
Network dropped middle of a frame
Resync complete
Waiting for IDR frame
Waiting for IDR frame
Discarding RTP packet queued for too long
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Waiting for IDR frame
Resync complete
Waiting for IDR frame
Can't empty video buffer
pi@raspberrypi ~ $

This is over Cat5 between the PC and the Pi2. Anyone seen this before?

darkpavian, 2015-07-14:
This looks similar to issue #153.

cgutman, 2015-11-23:
This is a duplicate of #198 and fixed in master.

closed 2015-11-24 by irtimmer