moonlight-stream/moonlight-embedded #169: "pair request refused" message when trying to pair pi with computer.

Tenket, 2015-07-16:
Hi all, i have gotten this working before I have all the required hardware and logged into steam and shield successfully and updated them both to latest version. 

But upon a reinstall of limelight 1.2.1 on the Pi B+ to try and fix some sound issues i find myself unable to make the thing pair anymore. 

More recently I have gotten the message "pair request refused"
Also before it was "error 401" with the verbose test saying "This client is not authorized. Certificate verification failed. (Error Code: 401)"
^No idea what changed between. I did literally nothing to either environment. 

Have any other people had issues with this?

cgutman, 2015-07-16:
The old Java client doesn't work with newer GFE versions. You'll have to use the new C client. Instructions for installing the package are in the readme.

closed 2015-07-16 by irtimmer

WilliamStam, 2015-07-27:
the new C client doesnt work on openelec tho :(

Tenket, 2015-07-28:
I was using rasbian and eventually got it working