moonlight-stream/moonlight-embedded #170: Four controllers - Keyboard always takes input slot

adzlc, 2015-07-16:
Hi all,

Not sure if I am the only one with this issue; I have 4 Xbox controllers hooked via a Raspberry pi to /dev/input/js0-3 (I've written udev rules so they are always these 4) and for other emulators this works fine but not for moonlight. My problem is the last player is ALWAYS the keyboard; 1-3 players will be a controller (regardless of how many controllers I connect). I have found this to be the case for every game, including games I know I can play with 4 controllers locally.

I have tried using -input but it appears that requires /dev/input/event which I believe can be random.

Has anyone come up with a solution?

Is it possible for the -input code to be altered (or a new argument added) to accept JS inputs?

adzlc, 2015-07-16:
I've tried creating a udev rule for the events as well; it would seem that the first device is always failing when there are four. I am running (without carriage returns):

```
moonlight stream -mapping xbox.map -input /dev/input/event4 -mapping xbox.map -input 
/dev/input/event5 -mapping xbox.map -input /dev/input/event6 -mapping xbox.map -input 
/dev/input/event7 -app Steam <IP-HERE>
```
This produces an error 'failed to open device: /dev/input/event4' . I can run evtest on all of these event4-7 though suggesting my udev rule is generally working (although I think I have broken the keyboard heh).

Before attempting to set the inputs I was just running:

```
moonlight stream -mapping xbox.map  -app Steam <IP-HERE>
```

adzlc, 2015-07-19:
I've tested all of the games locally vs streamed and all appear to support 4 controllers over a keyboard which suggests to me the keyboard is perhaps overriding one of the inputs even if you specify 4. I don't particularly know C so am not 100% sure if this is true so its just a hunch.

Spirit3s, 2015-08-04:
Same problem

cgutman, 2016-02-19:
This is fixed in common-c by https://github.com/moonlight-stream/moonlight-common-c/commit/40478fc19b9e5e5f6e7e153f97cdb0d21b53573e

closed 2016-02-19 by irtimmer