moonlight-stream/moonlight-embedded #171: xbox360 wireless controller input not working in games

Bztvuy, 2015-07-17:
Hello :)

I'm using the latest version of moonlight-embedded as of July 15 2015 8:00 pm EST from apt-get on a Raspberry Pi B+ running the latest Raspbian and I can't get my x360 wireless pad to work in games.  

I launch the stream and I get to steam and I have no problem moving around in steam, going through options, launching games, etc.  All the buttons works.  But as soon as I launch a game, I can't do anything.  It's like the controller goes dead.  

The only thing I can do is press the special button to bring up the steam overlay which I can still navigate without a problem.  

I have tried 3 different configurations including using xpad, xboxdrv as a daemon or launched in the terminal.  Either way, I know my configs work because I can navigate steam without any issue, it's only the games that are problematic.  

By the way, I could stream and play games 2 days ago without an issue so I suspect that the latest Geforce Experience, update which I applied yesterday on my pc (host), broke something.

I would like to know if anyone else is experiencing the same thing.

scoobyd00, 2015-07-17:
Have you tried the mapping option and created your own mapping file?

Bztvuy, 2015-07-17:
Yes, I've tried that with and without specifying which /dev/input to use.  Either way, I was getting unmapped button errors in the log when I didn't specify a mapping but not when I specified one so I guess that must be OK.

scoobyd00, 2015-07-17:
The last thing to try if you havent already is the init.script.
I have to use this for my wireless xbox controller to work correctly.
Have you tried  an init.d script with the daemon -D Option? 
Option 3 in this link
https://github.com/RetroPie/RetroPie-Setup/wiki/Setting-up-the-XBox360-controller

Bztvuy, 2015-07-17:
That's the one I'm using too.  Everything seems normal, it's only when I get in a game that nothing works anymore (but the keyboard still works).  evtest detects my controller correctly as event0 and xboxdrv detects it too.  The led is displaying player 1 like it should and all the buttons work in the stream menus.  I'm really at a loss here.

scoobyd00, 2015-07-17:
It does sound a very strange one, sorry I can't help you anymore,  I'm at a loss too, the one last thing you could try is to ignore steam and to try and launch the game directly with the -app option making sure you've added the game to the game stream option in geforce experience. 
This is the way I run games bypassing steam all together

Bztvuy, 2015-07-17:
I haven't tried that yet. I'll give it a shot later.  Thank you.

7hunderbug, 2015-07-17:
Because your problem has only started since updating GFE, I would install all NVidia software again from scratch using the most recent installer, do the GFE update afterwards and also reinstall Steam. When doing the Nvidia driver package install, choose "Custom Install" rather than "Express Install" so that you can check the box that does a clean install. Also, make sure you re-install Steam over the top of your already installed Steam otherwise you may end up deleting all your game files if you choose to remove Steam first (I learned that the hard way). Also, after the installs, reboot your PC, even if Windows does not insist on a reboot.
Also, check whether your controller is detected as an XInput device in Steam Big Picture Mode controller settings and make sure you don't have any other controllers connected to your PC.

Bztvuy, 2015-07-17:
You might be on to something here.  Steam is saying that it's an xinput device but I do have another controller receiver plugged into my host which is another xinput controller.  That must be it.  I'll try unplugging that.

mdjones, 2015-08-25:
I think I am having the same issue. I am using the XBOX wired controller directly connected to the host computer for testing. Everything works fine in Steam Big Picture navigation. 

Ex 1: Bioshock Infinite - If I launch app and play with host keyboard everything is fine. If I play with host wired XBOX controller there is no input detected. 

Ex 2: Borderlands: The Pre-Sequel - Host keyboard fine. Host wired XBOX controller causes crash with an image pop up window with several lines indicating "file not found"
 for the app exe

Hand copied one line since I could not locate the log. 
AK::Monitor::PostCode() Address = 0x217276d (filename not found) [in C:\Program Files (x86)\Steam\steamapps\common\BorderlandsPreSequel\Binaries\Win32\BorderlandsPreSequel.exe]

[image]

T3naciousC, 2015-09-24:
I have the exact same issue as mdjones...

It really confuses me why the xbox 360 controller works flawlessly while in games on the host but stops working after you start streaming the game (with the controller still connected to the host). Keybord works fine tho...
In my opinion this is something that is probably will need to get fixed at Nvidia side and is not a moonlight issue.
Really is a showstopper this, as the controller is not recognized as nicely via xboxdrv on the Raspberry pi client as it is on steam. 
To give an example, the game layers of fear, when controlling this with the 360 gamepad it plays like a charm, start streaming and this steam gamepad support is gone and the game gets unplayable, even tho you can get the buttons to map via xboxdrv.

Anyone found a solution to get the xbox 360 controller to just continue functioning while streaming as it functions when not streaming? (connected to host)

(FYI: I prefer keeping the controller connected to the host as this limits the amount of lag and the 360 controller has a very very good reception, even at longer distances)

T3naciousC, 2015-09-24:
Ok, that was easy to solve, from xda post

Using a wired/wireless controller connected to the GFE PC instead of the streaming device
Normally, Moonlight sends controller input from the streaming client which gets sent to the game by GFE. If you want to connect a controller to your PC instead of the streaming device, GFE can cause some problems because the emulated controller still appears to games as controller 1. Luckily there is a workaround for this. You'll need to rename the DLL that Nvidia is using to send controller input so it won't be used anymore. On 32-bit and 64-bit Windows, rename rxinput.dll to rxinput.dll.old on in C:\Program Files\NVIDIA Corporation\NvStreamSrv. On 64-bit versions, there's another copy of the DLL in C:\Program Files (x86)\NVIDIA Corporation\NvStreamSrv which you'll want to rename. You may have to do the renaming again if GFE does an update, but it should allow you to use your controller normally on games that only support 1 controller.

LaLeLiLeyLu, 2015-10-03:
I have the same issue here.

Everything works fine in multiplayer (local co-op) games but in singleplayer games the controller does not work. Using xbox360 wireless gamepads with xbox360 wireless hub on host pc.

Tried the workaround @T3naciousC but now I have no "Shield" tab in (V. 2.5.14.5) GFE -> Settings section anymore and moonlight stream does not work anylonger. I can´t pair or list again. I tried to rename the dlls back to origin name (+ restarting everything) but its still down. 

Any suggestions? Re-Install GFE?

LinkEscapist, 2015-10-10:
why dont u just plug the dongle 2 ur source pc?the wire's long enough,signal stable,ms originally support,

TomTsunami, 2015-10-25:
I am having the exact same problem.

Using a XBOX360 Wireless controller that is connected to the Host PC.  I can navigate Big Picture and I can press the Xbox button and it will take me out of whatever game I'm paying and show the steam overlay.  Interestingly, in Rocket League it prompts to press the Start button to Join the party (like a second player or something and it will add another player to my party)  But no other buttons work in game.

mdjones, 2015-11-09:
Hi All, 

Lost track of this but wanted you to know that I now have moonlight-embedded running great and use it routinely with my XBOX 360 Remote. I don't try and debug controller issues from the server anymore so not sure what was up with that. 

Launch on raspberry pi
$ moonlight stream 192.168.1.100 -mapping /home/pi/Moonlight/xbox.map -1080 -60fps -app Steam

Where xbox map is: 

 $ cat /home/pi/Moonlight/xbox.map
abs_deadzone = 0
abs_dpad_x = -1
abs_dpad_y = -1
abs_rx = 3
abs_ry = 4
abs_rz = -1
abs_x = 0
abs_y = 1
abs_z = -1
btn_dpad_down = 295
btn_dpad_left = 296
btn_dpad_right = 297
btn_dpad_up = 294
btn_east = 305
btn_mode = 316
btn_north = 308
btn_select = 314
btn_south = 304
btn_start = 315
btn_thumbl = 317
btn_thumbr = 318
btn_tl = 310
btn_tl2 = 312
btn_tr = 311
btn_tr2 = 313
btn_west = 307
reverse_dpad_x = false
reverse_dpad_y = false
reverse_rx = false
reverse_ry = true
reverse_x = false
reverse_y = true

I also set up the xbox controller as per: https://github.com/RetroPie/RetroPie-Setup/wiki/Setting-up-the-XBox360-controller

Server Specs:

Windows 10 Home
Intel(R) Core(TM) i7-2600 CPU @ 3.40GHz, 3701 Mhz, 4 Core(s), 8 Logical 
4 GB Ram
GTX 960

Client: Raspberry Pi 2 Model B
Network: Cat6E with Linksys E2500 router

closed 2016-01-16 by irtimmer

kriegalex, 2016-11-05:
I do have the same problem as OP and I don't think this is fixed....

T3naciousC wrote : "Normally, Moonlight sends controller input from the streaming client which gets sent to the game by GFE. If you want to connect a controller to your PC instead of the streaming device, GFE can cause some problems because the emulated controller still appears to games as controller 1".

I think this is exactly the problem. I do see two controllers in Steam Big Picture settings and the first one must certainly be some emulated stuff from GFE. When I try to test controller with Big Picture, only selecting second one makes my controller vibrate. So when I launch my game, it takes the first 1, which doesn't work because in fact the Xbox 360 controller is controller 2.

Now we have two choices :

- Delete the controller 1
- Set controller 2 as default

I haven't found how to do this sadly

kriegalex, 2016-11-05:
I had the same problem, tried so much things, controller is now plugged directly to the RBY... Not worth spending hours and hours just to avoid unplugging the USB cable