moonlight-stream/moonlight-embedded #175: Alsa error from writei: -11

poker-daddy, 2015-07-24:
Currently working from an Nvidia gtx 660 card, streaming successfully to a raspberry pi 2 model b.

Things seems to be fine; however, i am consistently encountering audio jitter and loops. upon exiting the stream, the error code encountered is "Alsa error from writei: - 11"

"-11" is the only error code i receive. My google searches have yielded naught but programming jargon. Is this something I can fix, or can potentially fix based on a setting/config option?

Kalagon, 2015-07-25:
As said by @cgutman in another thread:
> The Alsa errors are sometimes normal. The GFE will stop sending audio packets if there's no audio being played from the PC. This can cause underruns that lead to those messages in the logs.

Thus the Error might not be related to the jitter

LordFahrenheit, 2015-07-26:
Also having this problem, but I have frame stuttering on busy frames and absolutely no audio through HDMI.

Kalagon, 2015-07-26:
on my Pi2 I had to put
> hdmi_force_edid_audio=1
in my config.txt to get HDMI audio working

LordFahrenheit, 2015-07-26:
Still no audio, however, I turned on my speakers on the host computer so i get local audio there, can't figure out why it's not sending through stream though, -localaudio is not part of my command

adararnon, 2015-08-20:
Also having this "Alsa error from writei: -11" error.
The stream is REALLY laggy but I'm not having audio problems like you.
This happens on both 1080p and 720p (both 60 and 30 fps). In the 720p case, the lag is much worse for some reason.
I'm using moonlight-embedded 2.1.1, Windows 10, GFE 2.1.2 and nVidia driver 355.60.

vinanrra, 2015-08-29:
i have same alsa errors and are spaming my console :-1:

Batrosflow, 2015-09-09:
Hi there,
After going "crazy" trying all possible options, I found a good post explaining how to enable the audio on the RPi, I hope it helps you (I am not sure if the first 3 are required, but this is the order I followed):

	sudo apt-get install alsa-utils
	sudo apt-get install mpg321
	sudo apt-get install lame

	sudo modprobe snd-bcm2835

Sources:
https://jeffskinnerbox.wordpress.com/2012/11/15/getting-audio-out-working-on-the-raspberry-pi/
https://github.com/irtimmer/moonlight-embedded/issues/25

Please note that I am using:
- OSMC Release 2015.08-1
- Moonlight-embedded 2.1.1
- RPi 2

irtimmer, 2016-01-01:
Duplicate of #126

closed 2016-01-01 by irtimmer