moonlight-stream/moonlight-embedded #179: Loss Stats: Transaction Failed: 32

LordFahrenheit, 2015-07-28:
I'm using a Pi 2 overclocked and with 512 Gpu_mem, and a windows 7 desktop with a GTX 660 GPU. Both are connected through wifi (usb adapter on Pi and Adapter Card in Desktop) due to my Uni's troublesome internet policies. The video feed gets extremely choppy and freezes for significant amounts of time under heavy/busy frames; however I'm partially sure this may be a network issue because of the unreliability and traffic spikes due to the semi public nature of the wifi (running Iperf right now at 2 a.m. I get speeds of ~2Gbits but during the day it drops to an unreliable 8Mbits). The video stream however used to work fine except for the audio always was local to my computer regardless of commands. Anyway I digress from these issues on to the topic at hand... In an attempt to fix the audio problem I updated my Realtek audio drivers and it fully broke the streaming, so I rolled them back and now I'm having this issue:

Output on Pi:
pi@Moonlight ~ $ moonlight stream -app GoogleChrome [IP Address]
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Loss Stats: Transaction failed: 32
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...Segmentation fault

The Pi sets up everything but never actually displays the stream then hard quits, I'm pretty sure this is an audio issue considering I just changed the drivers AND when I add -localaudio it fixes the problem to the previous stuttering video with audio from my pc. I've tried defining -720 and -30fps but these have no effect on the video or the audio as they are the defaults anyway. I've also tried Steam and multiple games besides just regular chrome. I can post example pi output of stream with -localaudio if needed. Loss Stats: Transaction 32 is the primary error here, If i knew where Nvidia sent the audio to stream I think this would be easier to fix.

LordFahrenheit, 2015-07-30:
To anyone having this error this is what fixed for me: 1. Rollback Nvidia virtual Audio driver then when updating nvidia drivers through geforce experience select clean install.

irtimmer, 2015-08-08:
Patches have been merged so problem should be solved in the next release.

closed 2015-08-08 by irtimmer

Gehilfe, 2016-01-09:
Same problem again since yesterday (2016-01-10) :(
I already tried to reinstall nvidia drivers but without success.
Running moonlight-embedded on rpi2 (osmc)

FLo481, 2016-01-10:
I have the same problem. It didnt work with jessie and with wheezy neither. A complete reinstallation of the geforce driver + geforce experience was also unsuccessful. ( I am using moonlight-embedded from the repo, on a RPi 2)

donfalcone, 2016-01-13:
I have the same problem also. My setup is wheezy on rpi 2 and a nvidia 960. The problem appeard after seuil reinstall of the pi and an update of nvidia driver.