moonlight-stream/moonlight-embedded #180: Not working on Windows 10

scoobyd00, 2015-07-30:
Upgraded my main PC to Windows 10, latest geforce drivers installed 353.62 and Geforce experience 2.5.12.11 and the stream doesnt work.

Get the following error

Starting RTSP handshake...RTSP DESCRIBE request failed:500 
failed: -1

It sucessfully paired with the PC but the stream wont work?.

scoobyd00, 2015-07-31:
Windows firewall causing the issue.. fixed.

scoobyd00, 2015-07-31:
Ive spent all day trying to get moonlight to stream under windows 10 and failed.
 The stream works but its very very very laggy, tried different drivers etc etc.
 I did use some old GeForce drivers and got the stream to work without issues but it automatically updated GeForce experience and then caused the same laggy stream.

Ive eventually reinstalled Windows 8.1 and using the latest Geforce experience and NVidia drivers and it streams perfectly so it must be a issue with windows 10.

What was strange is that under windows 10 it streamed fine to my tablet.

Hopefully someone else can also do some testing and see what the issue is.

For now im sticking with windows 8.1 until we can confirm a working solution on windows 10

adararnon, 2015-08-01:
I'm suffering from the same problem. Problems started after I upgraded to Windows 10 and updated GFE. Both my PC and my RPi are connected by wire to the router so it can't be a network problem.

cgutman, 2015-08-02:
@scoobyd00 @adararnon Assuming you're having video choppiness/lag problems, I have an idea about what the issue may be. Are you testing on a Pi 1 or Pi 2? I'll post in an hour or so with a branch for you to try.

cgutman, 2015-08-02:
OK, clone https://github.com/cgutman/moonlight-embedded.git and checkout SpsFixupFix.

git clone https://github.com/cgutman/moonlight-embedded.git
cd moonlight-embedded
git checkout SpsFixupFix

joe772, 2015-08-03:
@cgutman: Seeing somewhat similar issues on Pi2. After a Windows 10 + Geforce 2.5 upgrade, I'm unable to stream smoothly on 1080/60fps (but just fine on 30fps). Seeing a ton of network issues, where I know it actually isn't:
```
Resync complete
Network dropped an entire frame
Resync complete
Network dropped an entire frame
Resync complete
Network dropped an entire frame
Resync complete
Network dropped an entire frame
```
I currently have moonlight installed from a repo (http://archive.itimmer.nl/raspbian/moonlight wheezy main), but have gone ahead and cloned the source branch you posted into a new directory. Not sure what exactly to do now though, as it's telling me there is no valid make file to use when I attempt to make it.

joe772, 2015-08-03:
Scratch that. Totally just finished reading through the README.

fehlis, 2015-08-03:
@cgutman Seeing the exact same issue as @joe772 on my PI2 here.
I did compile your version from the SpsFixupFix branch as you suggested, but sadly it doesn't change the behavior. Still very slow and tons of network messages.

scoobyd00, 2015-08-03:
Still not working for me either with that branch, still get the handshake error 500.
The very odd time I can get the stream to start its super slow with constant network errors.
This is on a pi2. It's perfect on 8.1

Ilos, 2015-08-03:
Have the same issue. Did a clean install on Win10.
Tried streaming to my Pi2 and nothing works, it stucks at 
>Initializing platform...done
>Resolving host name...done
>Starting RTSP handshake...^C

If i turn of Windows Firewall than i got this error:
>Initializing platform...done
>Resolving host name...done
>Starting RTSP handshake...connect() failed: 111
>failed: 111

It worked really good on Win 8.1, but now i cant get anything.
Even with your pull request @cgutman, no luck.
I tried to look, what the error 111 means, but for now i did not find anything :p

scoobyd00, 2015-08-04:
Looks like the handshake error may be down to the nvidia drivers.
Not sure what's causing the chopy video and all the network errors, its definitely not my network so either an issue again with the nvidia drivers or an issue with moonlight.

panfri787, 2015-08-04:
I have got the same issue that @joe772 and @fehlis too

Thanks for the support.

scoobyd00, 2015-08-04:
Looking over the net it seems this may be nvidia driver problem with windows 10.
People are complaining on the nvidia forums that streaming isn't working and other people that can get it to work have really poor choppy video.
Just like we are experiencing, so maybe not a issue with moonlight? But a driver problem?

panfri787, 2015-08-04:
If I select 60fps either 1080p or 720p the stream was laggy and moonlight says is losing frames by the terminal. If I select 30fps works well. Maybe it's a driver problem as you say. 

Thanks

poker-daddy, 2015-08-05:
Has anyone tried an older version of GFE on Windows 10? I just re-installed everything on Windows 8, including upgrading GFE to 2.5.1.12. That resulted in super choppy (and downright unusable) streaming. However, installing the older 2.4.5.28 completely fixed the issue.

scoobyd00, 2015-08-05:
I'm currently running windows 8.1 clean install with the latest 2.5.1.12 and mines smooth, only have an issue with windows 10

GigagamOr, 2015-08-07:
It seems to be a problem with the nvidia drivers rather than moonlight.

scoobyd00, 2015-08-08:
I see @cgutman  made some changes to moonlight-common the other day.
Has anyone tried a rebuild to see if the issues are fixed?

cgutman, 2015-08-08:
Those changes won't fix it. They are a fix for a deadlock that was already worked around in all moonlight-common-c clients and a fix for video breaking after 5 hours of continuous use.

scoobyd00, 2015-08-08:
Thanks for the update @cgutman . Do you know what the issue is?, or is it down to the nvidia drivers?

cgutman, 2015-08-08:
I can't reproduce the issue with my setup running Windows 10. If it's reproing in Win10 but not 8.1 with the same GFE version, it really seems like it must be somehow related to Win10.

scoobyd00, 2015-08-10:
Has anyone else have issues with windows 10 or any idea how we can sort out the serious lagging and network issues?

Ilos, 2015-08-10:
With my RPi2 and Win10-PC streaming is essentially fine for me.
My specs:
>RPi2 (overclocked with the RPi2 preset, Debian Jessie)
>PC (4790k, GTX 780, Win 10 home)
>Both are wired connected: PC <-> AVM Fritzbox 7490 <-> Rpi.

Well, it works fine if i set the bitrate <3500.
If i set the bitrate greater than ~3500 or keep it at standard bitrate then the stream starts to stutter.
With <3500 i can do playable(!) 720/60fps or 1080/30fps.
At 1080/60fps i need a bitrate around ~1500. But then i got an unplayable inputlag and the imagequality becomes unbareable.
But no matter what setting I use, I still got the warning
>Resync complete
>Network dropped an entire frame
But i really dont notice it ingame.

scoobyd00, 2015-08-10:
Theres defiantly an issue as we shouldnt be getting the dropped frames and bitrate at 3500 o below is very low, should be able to stream at standard bitrate.
Hopefully @irtimmer will be able to give us an idea if its moonlight or driver related

Ilos, 2015-08-10:
I really try to get into the code for days/weeks now, but unfortantely i'm still in the process of learning the language itself so i only make small progress.
I hope there is some solution for this issue.

irtimmer, 2015-08-10:
I'm trying to look into the Windows 10 problem, but for better debugging the problem it would be nice to have the C:\ProgramData\NVIDIA Corporation\NvStream\ logs for non lagging Windows 8 and/or non lagging Windows 10.

Ilos, 2015-08-11:
I have created 2 different Nv-logfiles.
One with standard arguments where the stream is stuttering unplayable.
And one where i set "-720 -60fps -bitrate 3500". I hope this helps in any way:
[no arguments](https://gist.github.com/Ilos/c276b5503b70af6fcba3)
[-720 -60fps -bitrate 3500](https://gist.github.com/Ilos/d063c5eb7494c21c4f16)

scoobyd00, 2015-08-11:
Sorry @irtimmer i couldnt provide and logs for window 10 as ive gone back to windows 8.1 until the issue is fixed. Ive looked but couldnt find any log files in my nvidia directory.

Luckily @Ilos has provided them :)

Looking at the 'windows 10' log it seems there are lots and lots of 

Obtaining SPS/PPS header
#3(I)[2015-08-11 06:35:20,333]{0000238C} averageBitrate 9670000, peakBitrate 9670000, vbvSize 169224, vbvInitialDelay 169224
#4(I)[2015-08-11 06:35:20,371]{00000D54} Received TCP frame invalidation range for stream 0: 0->1048575. Timestamp: 118253.288477
#5(I)[2015-08-11 06:35:20,748]{00000CD8} Frame 263 invalidates 249 250 251 252 253 254 255 256 257 258 259 260 261 262 263 264 265 266 267 268 269 270 271 272 273 274 275 276 277 278 279 280 281 282 283 284 285 286 287 288 289 290 291 292 293 294 295 296 297 298 299 300 301 302 303 304 305 306 307 308 309 310 311 312 313 314 315 316 317 318 319 320 321 322 323 324 325 326 327 328 329 330 331 332 333 334 335 336 337 338 339 340 341 342 343 344 345 346 347 348 349 350 351 352 353 354 355 356 357 358 359 360 361 362 363 364 365 366 367 368 369 370 371 372 373 374 375 376 377 378 379 380 381 382 383 384 385 386 387 388 389 390 391 392 393 394 395 396 397 398 399 400 401 402 403 404 405 406 407 408 409 410 411 412 413 414 415 416 417 418 419 420 421 422 423 424 425 426 427 428 429 430 431 432 433 434 435 436 437 438 439 440 441 442 443 444 445 446 447 448 449 450 451 452 453 454 455 456 457 458 459 460 461 462 463 464 465 466 467 468 469 470 471 472 473 474 475 476 477 478 479 480 481 482 483 484 485 486 487 488 489 490 491 492 493 494 495 496 497 498 499 500 501 502 503 504 505 506 507 508 509 510 511 512 513 514 515 516 517 518 519 520 521 522 523 524 525 526 527 528 529 530 531 532 533 534 535 536 537 538 539 540 541 542 543 544 545 546 547 548 549 550 551 552 553 554 555 556 557 558 559 560 561 562 563 564 565 566 567 568 569 570 571 572 573 574 575 576 577 578 579 580 581 582 583 584 585 586 587 588 589 590 591 592 593 594 595 596 597 598 599 600 601 602 603 604 605 606 607 608 609 610 611 612 613 614 615 616 617 618 619 620 621 622 623 624 625 626 627 628 629 630 631 632 633 634 635 636 637 638 639 640 641 642 643 644 645 646 647 648 649 650 651 652 653 654 655 656 657 658 659 660 661 662 663 664 665 666 667 668 669 670 671 672 673 674 675 676 677 678 679 680 681 682 683 684 685 686 687 688 689 690 691 692 693 694 695 696 697 698 699 700 701 702 703 704 705 706 707 708 709 710 711 712 713 714 715 716 717 718 719 720 721 722 723 724 725 726 727 728 729 730 731 732 733 734 735 736 737 738 739 740 741 742 743 744 745 746 747 748 749 750 751 752 753 754 755 756 757 758 759 760 761 762 763 764 765 766 767 768 769 770 771 772 773 774 775 776 777 778 779 780 781 782 783 784 785 786 787 788 789 790 791 792 793 794 795 796 797 798 799 800 801 802 803 804 805 806 807 808 809 810 811 812 813 814 815 816 817 818 819 820 821 822 823 824 825 826 827 828 829 830 831 832 833 834 835 836 837 838 839 840 841 842 843 844 845 846 847 848 849 850 851 852 853 854 855 856 857 858 859 860 861 862 863 864 865 866 867 868 869 870 871 872 873 874 875 876 877 878 879 880 881 882 883 884 885 886 887 888 889 890 891 892 893 894 895 896 897 898 899 900 901 902 903 904 905 906 907 908 909 910 911 912 913 914 915 916 917 918 919 920 921 922 923 924 925 926 927 928 929 930 931 932 933 934 935 936 937 938 939 940 941 942 943 944 945 946 947 948 949 950 951 952 953 954 955 956 957 958 959 960 961 962 963 964 965 966 967 968 969 970 971 972 973 974 975 976 977 978 979 980 981 982 983 984 985 986 987 988 989 990 991 992 993 994 995 996 997 998 999 1000 1001 1002 1003 1004 1005 1006 1007 1008 1009 1010 1011 1012 1013 1014 1015 1016 1017 1018 1019 1020 1021 1022 1023 1024 1025 1026 1027 1028 1029 1030 1031 1032 1033 1034 1035 1036 1037 1038 1039 1040 1041 1042 1043 1044 1045 1046 1047 1048 1049 1050 1051 1052 1053 1054 1055 1056 1057 1058 1059 1060 1061 1062 1063 1064 1065 1066 1067 1068 1069 1070 1071 1072 1073 1074 1075 1076 1077 1078 1079 1080 1081 1082 1083 1084 1085 1086 1087 1088 1089 1090 1091 1092 1093 1094 1095 1096 1097 1098 1099 1100 1101 1102 1103 1104 1105 1106 1107 1108 1109 1110 1111 1112 1113 1114 1115 1116 1117 1118 1119 1120 1121 1122 1123 1124 1125 1126 1127 1128 1129 1130 1131 1132 1133 1134 1135 1136 1137 1138 1139 1140 1141 1142 1143 1144 1145 1146 1147 1148 1149 1150 1151 1152 1153 1154 1155 1156 1157 1158 1159 1160 1161 1162 1163 1164 1165 1166 1167 1168 1169 1170 1171 1172 1173 1174 1175 1176 1177 1178 1179 1180 1181 1182 1183 1184 1185 1186 1187 1188 1189 1190 1191 1192 1193 1194 1195 1196 1197 1198 1199 1200 1201 1202 1203 1204 120
#6(I)[2015-08-11 06:35:20,763]{0000238C} Frame 263 is an I-Frame.

errors that dont appear in the non stuttering -bitrate 3500

irtimmer, 2015-08-11:
It looks like the problem is in the recovery of dropped packets. By lowering the bitrate the chances on dropped packets is lowered and it won't stutter. But I'm not sure if this is caused by Windows 10 dropping more packets or if it is has problem recovering. In my logs it look like Windows 10 skipped multiple frames before providing an IDR, but maybe Windows 8 did the same.

cgutman, 2015-08-11:
Those logs don't indicate the problem. They are a symptom of it. The lost packets reported by Moonlight cause an IDR frame to be requested, leading to the I-frame related logging. The frame invalidation is due to the way current Moonlight clients work. Clients built with updated common libs will not use frame invalidation. There won't be any change to the stuttering from the new change though.

Ilos, 2015-08-12:
I found something interessting, but i don't know what to think about it:
So i experimented with the bitrate and something strange happened:
As i said above, if i initiate the stream with: 
>moonlight stream 192.168.0.5 -app "GAME" -720 -60fps -bitrate 3500 -mapping XBONE.map

everything works without stuttering or lag.
So i wanted to look at witch bitrate the stream starts to stutter and i got it as high as 7500
>moonlight stream 192.168.0.5 -app "GAME" -720 -60fps -bitrate 7500 -mapping XBONE.map

At 7500 the stream works fine without any stutters.
So, I wanted to see, if i can get higher bitrate if i use 30fps.
But strange is, i couldn't even get 7500. The highest bitrate, that is "playable", is ~4000.
Everything above produces a stuttering, unplayable stream.

tl;dr:
-720 -60fps -bitrate 7500 ----> playable, little to no suttering
-720 -30fps (-bitrate 5000) --> not playable, heavy stuttering
-720 -30fps -bitrate 4000 ----> still playbale, litte stuttering.
So if you want a playable stream you should prefer 60fps over 30fps, but for me this is not logical.

scoobyd00, 2015-08-13:
Has anyone tried to see if the latest 355.60 drivers fix anything?

fehlis, 2015-08-13:
I installed it earlier today and my problems persist sadly.
I tried the following settings as I did with the former Driver-Version:
1080p / 60fps / 7500kbit  -> slideshow
1080p / 60fps / 3500kbit  -> slideshow (and crappy quality, obviously)
1080p / 30fps / 7500kbit  -> works fine

tseruga, 2015-08-14:
Having the same issues.

Using a Raspberry Pi 1 running Raspbian with a host machine running Windows 10. Almost all settings are slideshows except for 720p / 30fps / 7500kbps.

Any word as to who's end this bug is on?

cgutman, 2015-08-15:
Can you still reproduce the issue with Moonlight Embedded 2.1?

scoobyd00, 2015-08-15:
Hopefully @Ilos will be able to test for us. Im currently running 8.1 because of the issue with moonlight and so havent go back to windows 10 yet until we can confirm its fixed.

fehlis, 2015-08-15:
I just built v2.1.1 on my Pi2 and for me, streaming works again smoothly with Windows 10!
I'm running GF Exp 2.5.12.11 , Driver Version 355.60

I tested with:
1080p / 60fps / 7500kbit -> smooth playback

scoobyd00, 2015-08-15:
Excellent. Im going for a windows 10 install now :)

scoobyd00, 2015-08-15:
I cant get the stream working on windows 10.

Ive paired succesfully but keep on getting the 

Starting RTSP handshake...RTSP DESCRIBE request failed:500 
failed: -1  error

Ive tried reinstalling drivers and experience but no luck.

Anyone have anymore ideas?

scoobyd00, 2015-08-15:
Further update.

The stream works fine on my android tablet and windows 10.

So it looks like its an issue with moonlight-embedded?.

fehlis, 2015-08-15:
Hmm strange. Im using not using -app. Just default steam. I had to re-pair first but no issues afterwards.
If there's any info i can provide to narrow this down, let me know.

scoobyd00, 2015-08-15:
Thanks. I dont know whats happened but the stream has started to work.
I used it on the android tablet a couple of times and then tried the pi and it started working.
How I dont know!!.. ill see how it goes over the next couple of days.

joe772, 2015-08-17:
2.1 + latest windows 10 driver and things look to be running smoothly now on w10. Thanks guys!

tseruga, 2015-08-17:
Things seem to be working for me as well.

closed 2015-08-17 by irtimmer