moonlight-stream/moonlight-embedded #182: Network dropped an entire frame

scoobyd00, 2015-07-31:
Thought I had the stream working on windows 10, played one game, turned off pc and when I tried again the video is very very choppy and unplayable.

I'm getting the below errors.
I know its not a network issue over the router as its streaming fine to my tablet.

Any ideas, ive tried everything and cant get it working.
I'm sure I had this issue before when I was on windows 8.1 and it was something to do with the audio drivers on the pc but ive uninstalled all my drivers and its still not working.

Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...Network dropped beginning of a frame
Waiting for IDR frame
Resync complete
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
done
Starting input stream...done
Alsa shortm write, write 20 frames
Alsa error from writei: -11
Alsa error from writei: -11
Alsa error from writei: -11
Network dropped an entire frame
Resync complete
Network dropped an entire frame
Resync complete
Network dropped an entire frame
Resync complete
Network dropped an entire frame
Resync complete
Network dropped an entire frame
Resync complete
Network dropped an entire frame
Resync complete
Network dropped an entire frame
Network dropped an entire frame
Network dropped an entire frame
Network dropped an entire frame
Resync complete
Network dropped an entire frame
ame
Waiting for IDR frame
Resync complete
Waiting for IDR frame
Waiting for IDR frame
Alsa error from writei: -32
Loss Stats: Transaction failed: 32
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Terminating RTSP handshake...done

scoobyd00, 2015-07-31:
This issue is related to Windows 10

Ive spent all day trying to get moonlight to stream under windows 10 and failed.
The stream works but its very very very laggy, tried different drivers etc etc.
I did use some old GeForce drivers and got the stream to work without issues but it automatically updated GeForce experience and then caused the same laggy stream.

Ive eventually reinstalled Windows 8.1 and using the latest Geforce experience and NVidia drivers and it streams perfectly so it must be a issue with windows 10.

What was strange is that under windows 10 it streamed fine to my tablet.

Hopefully someone else can also do some testing and see what the issue is.

For now im sticking with windows 8.1 until we can confirm a working solution on windows 10.

scoobyd00, 2015-08-01:
Same issue as not working on windows 10

closed 2015-08-01 by scoobyd00

demonLaMagra, 2015-09-19:
Same issue...

Pi 2 is overclocked and still terrible lag. Windows 10 is obviously the prime suspect! 

Any may have to revert back to 8.1 :(