moonlight-stream/moonlight-embedded #184: "Could NOT find Freescale" -- any help?

roboticaa, 2015-08-04:
"cmake ../" returns the following message:

-- Could NOT find Freescale (missing:  FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY)
-- Configuring done
CMake Error: Unable to open check cache file for write. /home/pi/moonlight-embedded-2.0/build/CMakeFiles/cmake.check_cache

Any ideas what I need to do to fix? I've tried completing the rest of the install but the "moonlight pair" command just segfaults.

darkpavian, 2015-08-11:
Judging from your home die I would say you are using some raspberry pi. 
I do not know exactly, but I would say that you do not need the Freescale library. I think it is only required for Freescale CPUs
http://www.freescale.com/webapp/sps/site/homepage.jsp?code=IMX_HOME

roboticaa, 2015-08-12:
Thanks, yeah its a raspi 2. Its the only error i get when i'm making the file set. When I try and run moonlight I get a segfault and can't think what else might be doing it.

To be clear, I've tried ignoring the freescale error but it still doesn't work.

darkpavian, 2015-08-13:
I don't think the missing freescale libs are responsible for this. 
Do you also get the seg fault when running as root (sudo moonlight...)?

roboticaa, 2015-08-13:
if I use 'moonlight' or 'sudo moonlight' it returns the list of available commands.
if I use 'moonlight pair' or 'sudo moonlight pair' it returns a seg fault. (using sudo gives the same return)

I can enter 'moonlight map' and 'moonlight help' and get a response but these don't help me pair the raspi to my desktop.

Do you think the moonlight installation needs fixing? I tried following the installation through again but that didn't seem to change anything.

Thanks for the help btw!

darkpavian, 2015-08-13:
It was just a guess, as i encountered seg faults due to rights problems when running something not as root. 
I think there is another cmake problem. Which cmake version are you using?
Are there any "Permission denied" errors after the "Unable to open check cache file" error?
Did you create the build directory as root ("sudo mkdir build")?
Maybe you could try to start over with a clean build directory, maybe in another dir. Just make sure that you don't create the dir as root.

closed 2015-09-12 by irtimmer