moonlight-stream/moonlight-embedded #186: Internet access, keyboard.h change, key combination Ctrl+Alt+Shift+Q

joh9, 2015-08-12:
Hello,

first thanks for the extreme cool software !

Data: OS = Raspbian, Me = Linux Noob, Moonlight = Installation Instructions -> Raspbian Wheezy (deb http: //archive.itimmer.nl/raspbian/moon ......), apt-get update
apt-get, install moonlight…..
Everything works great. Under Win10 it is laggy, but I found  Issue Solution Bitrate < 3500 ( https://github.com/irtimmer/moonlight-embedded/issues/180)

I have 3 questions.

- So I can access from the Internet to my home desktop PC, I've built a OpenVPN network. Is there any other way, such as in Android? ->
Ports to forward manually:
TCP 35043, 47984, 47989, 47995, 47996, 48010
UDP 47998, 47999, 48000

- Pressing the spacebar results in the number 2 key. I have the Issue Post https://github.com/irtimmer/moonlight-embedded/issues/155 found. How can I change / add the correct keyboard.h file?
I find keyboard.h only in “/usr/include/linux/keyboard.h”.

- The key combination Ctrl + Alt + Shift + Q do NOT exit streaming session, what I’m doing wrong? 

Many thanks for your help.

irtimmer, 2015-08-15:
The key combination and the spacebar fix are only available in the newly released versions. So update using apt-get update and apt-get upgrade and those should be fixed.

joh9, 2015-08-18:
Thanks a lot irtimmer . I had actually already performed the update , but was apparently just before the update release date , so I thought I would have the latest update :-) .

- In terms of the Internet access there is no way for moonlight ?

irtimmer, 2015-08-18:
It's possible to stream via the internet when ports are forwarded but I don't exactly understand what kind of functionality of the Android version you are missing in Moonlight Embedded.

joh9, 2015-08-18:
thx for your fast reply. Under "Quick Start" ( https://github.com/irtimmer/moonlight-embedded ) you wrote - Ensure your GFE server and client are on the same network. And i forwarded the Ports 
TCP 35043, 47984, 47989, 47995, 47996, 48010
UDP 47998, 47999, 48000
but the connection failed. 
I should try it again --> Sry, no reaction. I paired the PC over the Home network, but if i will stream or "moonlight list" nothing happend !?
If i want pair again with "moonlight pair X.X.X.X" nothing happend ....

joh9, 2015-08-18:
Arrggg, Sry wrong ip. --> "moonlight stream X.X.X.X -app .....
Big THX for your great work !!!

closed 2015-08-18 by irtimmer