moonlight-stream/moonlight-embedded #189: Bad version in Packages.gz

dodslaser, 2015-08-15:
The version number for moonlight-embedded in Packages.gz on the debian repo is 2.1.**0**-1 rather than 2.1.**1**-1. This causes apt-get to return a 404.

```shell
E: Failed to fetch http://archive.itimmer.nl/raspbian/moonlight/pool/main/m/moonlight-embedded/moonlight-embedded_2.1.0-1_armhf.deb  404  Not Found
```

scoobyd00, 2015-08-15:
Getting the same error here

irtimmer, 2015-08-15:
Try by rerun 'apt-get update' as there's nothing wrong with packages.gz. Propably this is just a cache issue.

closed 2015-08-15 by irtimmer

scoobyd00, 2015-08-15:
Ive dont multiple re runs of apt-get update but still get the same error as dodslaser.
Dont know if it takes time to sort the cache out but ive rebooted and re run apt-get update and still the same

scoobyd00, 2015-08-15:
Fixed now .. Took a while to re cache