moonlight-stream/moonlight-embedded #190: Platform 'default' not found

Ilos, 2015-08-15:
So with the newest commits i can't get the stream to start (or pair or anything).
If i start moonlight, then the message in the issue-title pops up.
Am i missing something? Is there something i forgot to install?
```
#ifdef HAVE_PI
  if (std || strcmp(name, "pi") == 0) {
     void *handle = dlopen("libmoonlight-pi.so", RTLD_NOW | RTLD_GLOBAL);
     if (handle != NULL && dlsym(RTLD_DEFAULT, "bcm_host_init") != NULL)
       return PI;
  }
#endif
```
It has to do with this code section in src/platform.c, right?
Well atleast i think, because the platform_check does not return PI to "enum platform system" in src/main.c

irtimmer, 2015-08-15:
Please provide some more information about used distribution, how installed, installation log etc. A single error message isn't very useful for debugging.

Ilos, 2015-08-15:
I have a Rpi2 with Debian wheezy (Raspbian), but i also tried a beta-version of Debian Jessie.
Both had the issue with your latest package you released.
Now i tried to build it from source and it worked without the issue.
I think this can be closed.

cgutman, 2015-08-15:
If you had the issue with the latest package, it's not fixed. This bug should be reopened. The packages should work.

scoobyd00, 2015-08-15:
Strange, I've used the latest package on a fresh install of raspbian and didn't get this issue, I've had other issues but not this one

irtimmer, 2015-08-15:
Issue should already be solved by the latest version. So only when someone has still problems with 2.1.1 this bug will be reopened.

closed 2015-08-15 by irtimmer

cgutman, 2015-08-15:
It does appear to be fixed with the latest package.

codewaffle, 2015-09-07:
Getting this in a Rasperry Pi Model B running Raspbian, everything's up to date, I've tried removing and reinstalling moonlight and related packages to no avail.

uname/apt/moonlight output: https://gist.github.com/codewaffle/d16ef4346a7ed58d07f5