moonlight-stream/moonlight-embedded #191: RTSP DESCRIBE request failed:500

scoobyd00, 2015-08-15:
Ok, ive gone for a new Windows 10 install .

Ive succesfully paired moonlight but when i come to run the stream i get the following error message.

Starting RTSP handshake...RTSP DESCRIBE request failed:500 
failed: -1

Ive disabled windows firewall, got static ip addresses for both my raspberry pi and pc but i still cant get the stream to work.

It worked fine in windows 8.1

Any ideas its driving me crazy.

Ilos, 2015-08-15:
try to reinstall your drivers and Geforce Experience.

irtimmer, 2015-08-15:
Duplicate of #180

closed 2015-08-15 by irtimmer

scoobyd00, 2015-08-15:
Still getting the Starting RTSP handshake...RTSP DESCRIBE request failed:500 
failed: -1 error
Tried re installing drivers and experience but the stream wont start. Pairing is fine.

Anyone know what the error 500 means?.