moonlight-stream/moonlight-embedded #192: Platform 'default' not found

Zega, 2015-08-15:
Hello everyone,

I had moonlight 2.1.0 installed on my Rasp 2 B, I am running windows 8.1.

The pairing was good, I tested the stream, it worked but was not that's good, before I tried to solve this I decided to install the update I found for moonlight (2.1.0-1).

After the update I have now the following error when I try to stream or repair : 
Platform 'default' not found

What does that mean? I am quite totally a noob on linux, raspberry, github, etc... sorry if it is an obvious problem!

Ilos, 2015-08-15:
#190

Zega, 2015-08-15:
@Ilos Thank you, which version do you build from source? The 2.1.1? Is there any (easy) "how-to" to do this? (as I was saying I do not know that much about linux).

Thank you.

Ilos, 2015-08-15:
To build from source you need to clone the git repository to your Raspberry and build it from their.
[Their is a small description in the Wiki](https://github.com/irtimmer/moonlight-embedded/wiki/Compilation)

```
cd
git clone https://github.com/irtimmer/moonlight-embedded
cd moonlight-embedded
```
and from here, follow whats written in the wiki

irtimmer, 2015-08-15:
Duplicate of #190

closed 2015-08-15 by irtimmer

Zega, 2015-08-15:
@Ilos Thank you for your help.

I have now a problem with cmake, I need the 3.1 or higher but I don't know how to install it, could you help me?

ObviousLeeGitHub, 2015-08-15:
@Zega I have exactly the same problem. Will be following this :)

ObviousLeeGitHub, 2015-08-15:
This comment from [SkewPL](https://github.com/SkewPL) seems to be the fix.. I'm installing Cmake 3.2.3 right now..:

@unchartedxx sent instructions how to compile new version of cmake here: #104 , that worked for me and i compiled moonlight without problems.

Download, compile and install cmake 3.2:

sudo apt-get install build-essential
wget http://www.cmake.org/files/v3.2/cmake-3.2.3.tar.gz
tar xzf cmake-3.2.3.tar.gz
cd cmake-3.2.3
./configure
sudo make install

Compile and install moonlight:

cd moonlight-embedded-2.0
mkdir build
cd build/
cmake -D CMAKE_BUILD_TYPE=Release ../
make
sudo make install

cgutman, 2015-08-15:
Building from source is not a fix! The package is broken and we need to figure out why.

ObviousLeeGitHub, 2015-08-15:
That would be right. I'm new to the scene ;-)

ObviousLeeGitHub, 2015-08-15:
I can't get it to work.. so still no temporary solution

 sudo cmake -D CMAKE_BUILD_TYPE=Release ../
 
-- Could NOT find Freescale (missing:  FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY)
-- Could NOT find CEC (missing:  CEC_INCLUDE_DIRS CEC_LIBRARIES) (Required is at least version "3.0")
-- checking for module 'sdl2>=2.0.4'
--   package 'sdl2>=2.0.4' not found
-- checking for module 'libavcodec'
--   package 'libavcodec' not found
-- checking for module 'libavutil'
--   package 'libavutil' not found
-- checking for module 'libswscale'
--   package 'libswscale' not found

cgutman, 2015-08-15:
The new 2.1.1 package resolves the issue.

Try:
sudo apt-get update

If you have moonlight-embedded installed then run:
sudo apt-get upgrade
(You should see it upgrade moonlight-embedded to 2.1.1)

Otherwise:
sudo apt-get install moonlight-embedded

miroky, 2015-08-15:
Let's try it! Thanks :3

miroky, 2015-08-15:
It doesn't work...

I follow the steps wrote up here, and still doesn't work.

That's my output in the command 'cmake -D CMAKE_BUILD_TYPE=Release ../'

-- The C compiler identification is GNU 4.6.3
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
CMake Error at /usr/local/share/cmake-3.2/Modules/FindPackageHandleStandardArgs.                                                                                       cmake:138 (message):
  Could NOT find ALSA (missing: ALSA_LIBRARY ALSA_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/local/share/cmake-3.2/Modules/FindPackageHandleStandardArgs.cmake:374 (_F                                                                                       PHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.2/Modules/FindALSA.cmake:56 (FIND_PACKAGE_HANDLE_STAN                                                                                       DARD_ARGS)
  CMakeLists.txt:16 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/moonlight-embedded-2.1.1/build/CMakeFiles/CMakeOutput.log".

Zega, 2015-08-15:
I totally remove moonlight and get the 2.1.1, it is now working again, no more "Platform default not found", but terrible input lag (video and sound are ok) and after 5 minutes I get the following errors:

IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame

etc...

until it crashes:
Can't empty video buffer

I do not know if it is an 2.1.1 issue or if it has something to do with my config.

cgutman, 2015-08-15:
@miroky Don't build, just run "moonlight" from the terminal