moonlight-stream/moonlight-embedded #194: Can't open configuration file: hosts/XX.XX.XX.XXX.conf

ilovemetrics, 2015-08-15:
I'm getting this error while trying to run: moonlight stream XX.XX.XX.XXX (local network IP of my desktop). I was able to successfully pair, but not stream. Firewalls have been disabled.

Any help would be appreciated.

ilovemetrics, 2015-08-15:
I'm running the latest version of moonlight embedded, btw.

Ilos, 2015-08-15:
I actually did a workaround to this issue as I have the same problem.
I just made an folder "hosts" and created a file named "XXX.XXX.XXX.XXX.conf" (of course with the IP of your host-pc).
You should then start moonlight directly outside the hosts-folder.
There should be a option to make the config-file optional, not necessary.

ilovemetrics, 2015-08-15:
Thanks for the response. Weirdly enough, I just got it working. Still getting the error, but it doesn't seem to prevent connection now. The only thing I did since posting was scan my computer for games in the GeForce Experience console. Not sure why that would have any effect, but it works like a charm now!

cgutman, 2015-08-15:
This message is harmless. It really should be made less "error-like" to avoid causing confusion.

closed 2015-08-15 by irtimmer