moonlight-stream/moonlight-embedded #196: Controller mapping not working

ilovemetrics, 2015-08-15:
Trying to map a Logitech F310 wired gamepad to moonlight embedded on a Raspberry Pi 2.

Running "moonlight map -input /dev/input/event0 controllers.map" starts execution, but just hangs there forever.

Any help is appreciated.

Ilos, 2015-08-15:
Can you run evtest? (You can install it if you don't have it: sudo apt-get install evtest).
What does it say?

ilovemetrics, 2015-08-15:
I had some luck running "moonlight map controllers.map" (not sure why you don't need to specify -input, etc.) but I wasn't able to complete all the button mappings correctly. It didn't detect the left/right bumper or special buttons. Ran moonlight again and was able to get the left stick working to navigate through menus, but nothing else worked.

I'll run evtest and post the results here. Thanks.

ilovemetrics, 2015-08-16:
All inputs responded correctly. Output below:

Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x46d product 0xc21d version 0x4014
Input device name: "Logitech Gamepad F310"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 304 (BTN_A)
    Event code 305 (BTN_B)
    Event code 307 (BTN_X)
    Event code 308 (BTN_Y)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value    128
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 1 (ABS_Y)
      Value   -129
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 2 (ABS_Z)
      Value      0
      Min        0
      Max      255
    Event code 3 (ABS_RX)
      Value    128
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 4 (ABS_RY)
      Value   -129
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 5 (ABS_RZ)
      Value      0
      Min        0
      Max      255
    Event code 16 (ABS_HAT0X)
      Value      0
      Min       -1
      Max        1
    Event code 17 (ABS_HAT0Y)
      Value      0
      Min       -1
      Max        1
  Event type 21 (EV_FF)
    Event code 80 (FF_RUMBLE)
    Event code 81 (FF_PERIODIC)
    Event code 88 (FF_SQUARE)
    Event code 89 (FF_TRIANGLE)
    Event code 90 (FF_SINE)
    Event code 96 (FF_GAIN)

ilovemetrics, 2015-08-16:
Think I figured it out. Wasn't clear from the docs on how to set this up, and there are lots of old instructions out there.

Resolved by creating the mapping file (moonlight map [filename].map) and then specifying the mapping file at launch (moonlight stream -mapping [filename].map XX.XX.XX.XXX).

Still having issues with mapping the left/right bumpers during setup, but will dig into that some more later.

Thanks for the help.

Ilos, 2015-08-16:
Hmm... It seems like your pad doesn't have bumper inputs. But hardware wise they are there.
Does the pad work on windows correctly?
Maybe try to edit the map-file manually.
Bumper-btns left and right with 310 and 311.
Trigger-btns should be both -1.
Therefore abs_z should be 2 and abs_rz 5

ilovemetrics, 2015-08-16:
Thanks, Ilos. I was able to map the file manually and got it working. For some reason I wasn't able to trigger a response using the map function, but all is well now and everything is working great.

Thanks again.

closed 2015-08-16 by ilovemetrics