moonlight-stream/moonlight-embedded #198: Crashes in 1080 on Win 10

Zega, 2015-08-16:
So I get a new issue :(

When I start to stream with "moonlight stream -1080" I got a terrible input lag, even if the video seems to played smoothly, and after a moment it crashes. This is the message I got:

'IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame'

etc...

until it crashes:
'Can't empty video buffer'

I tried with -bitrate but it is the same problem.

Whait is really strange is that when I use "moonlight stream -720 -bitrate 15000" everything is running realy smoothly!!
I think there is a problem with the 1080 argument, does anyone have the same problem?

I am running moonlight embedded 2.1.1 on windows 10.

Zega, 2015-08-16:
I tried again with -bitrate and with an higher value, video is running fine, audio is only 2.0, but it's still unplayable as the input lag is really really high! any clue?

cgutman, 2015-08-16:
If you're running on a Pi, you'll need to overclock your h264 decoder to support 1080p60 (which is what your set of flags specifies). If you don't want to, you can add the -30fps flag.

If you want to try overclocking, I run these settings (and ONLY these settings) on my Pi 2:
force_turbo=1
h264_freq=333

If you add more overclocking/overvoltage flags, you risk tripping the warranty bit.

Zega, 2015-08-17:
@cgutman I am using a Raspberry Pi 2 B, I shouldn't have to overclock it, no?

Now, video and sound are completly ok in 1080p, 60fps at highest bitrate, it plays very smoothly.
BUT I still have a terrible, terrible "input lag".

I am using a Xbox 360 controller plugged wirelessly to my PC.

To be more precise: when I click on a button: on my PC screen it reacts immediatly, on my Rasp/TV I have a delay of at least 1 sec even if everything is running smoothly on it.

And I do not have this issue at all on 720p 60fps highest bitrate.

PacoBarba, 2015-11-01:
Hello, i am new user in Git and speak spanish, sorry for my bad english.
i have the same problem with 1080 parameter and i am Raspberry pi 2.
With "-720 60fps -bitrate 30000" runs smootly on ethernet.
Witch "-1080 60fps -bitrate 30000" runs also smootly but in some part of some games moonlight breaks and show 'Can't empty video buffer'. Investigating, i observed that only fail in determinates area s in the some games.
An example in "Metal Gear Solid V" when  i use night vision moonlight breaks and show the empty buffer error. In the Fx Pinball 2 fails when i select table. After the fail relaunch the moonlight and runs fine, other games works perfectly all the time. 
A not know the origin of the problem but is very estrange.

PacoBarba, 2015-11-02:
I make these test in Windows 7 Pro 64 bits and Windows 10 Pro 64 bits

Rutherberg, 2015-11-20:
I have the exact same problem with Dirt 3 and NFS:HP so far. 
1080p60 (all other options are default) runs butter smooth with an acceptable input lag (Xbox360 wireless).
My RPi 2 runs Raspbian Jessie and is a little bit overclocked and cooled with a fan (max 36°C) : 

force_turbo=1
h264_freq=500
over_voltage=2
gpu_mem=256

As described, when there is a cinematic change, i.e. when I pass the finish line, moonlight crashes with "Can't empty video buffer".
I suspect a new version of the nVidia driver or GFE to be the cause because the problem wasn't there before.
My host PC runs Win7 (i7, 16Gb ram, GTX660Ti), with GFE 2.7.4.10 and drivers 358.91. All is connected via ethernet with a gigabit switch.
Any ideas or workaround ?

Rutherberg, 2015-11-21:
I did few more tests. I did a fresh install on RPi 2 (wheezy this time). Again, crashes when abrupt graphic changes. But I raised the gpu_mem from 256 to 512 with no other overclocking options and the phenomenon (almost) disappeared with Dirt 3 but not with NFS:HP. I said almost because when I change the pilot view too quickly in Dirt 3 it crashes, but anymore when I cross the final line.
Now I will test it with older versions of GFE and drivers.

cgutman, 2015-11-21:
I think we're not telling OMX to give us large enough buffers for the worst case frame size. I'm going to try to reproduce it here.

Rutherberg, 2015-11-21:
Ok, thanks !
I reconnected my RPi B with the exact same install (but overclocked), it runs @1080p30 with no crashes.
With RPi B, I noticed when there is abrupt graphic changes, moonlight didn't crash but I get some 

IDR frame request sent
Waiting for IDR frame

But not with the RPI 2, it just crashes.
Hope you can reproduce it, I would like to test it with RPi B+ but I ran out of SD cards :-P

cgutman, 2015-11-21:
So you're getting the crash on a Pi 2 running Jessie with your above overclock settings when streaming 1080p60 at the default bitrate? No other command line args?

Rutherberg, 2015-11-21:
I tested the RPi 2 with Wheezy and Jessie, both up-to-date, no differences.
It crashes in 1080p, 30 and 60fps. As mentionned, it's a little bit better with gpu_mem 512, but didn't fix the problem. Overclocked or not, the problem stays the same.
I launch "moonlight stream -1080 -30/60fps -mapping xxx.map xxx.xxx.xxx.xxx", all the rest is default.

Prakson, 2015-11-21:
I have very similar problem - Rpi2, tested wheezy and ArchLinuxArm.
I don't have any input lag, everything works well but after some time Moonlight exits with "Can't empty video buffer" message. 
As bitrate is bigger the moonlight works shorter, but I found scene in Witcher 3 when the moonlight crash every time. I tried three times - moonlight crashed on every attempt.

Rutherberg, 2015-11-21:
Yes, I think it's recurrent with heavy graphics. I just tried to crash it with other games like Trine, Block that matters, etc... but without succes, it stays up and running.

cgutman, 2015-11-21:
Heh, got a repro immediately (just starting Steam) when I made my host PC drop 0.1% of video packets. Second IDR frame killed it. The issue was what I thought: OMX gave us an input buffer of 81920 bytes and the frame data was 91486 bytes.

cgutman, 2015-11-21:
OK, I fixed it and opened a PR: https://github.com/irtimmer/moonlight-embedded/issues/246

You can try it on Jessie (or Arch) using my special branch:

git clone https://github.com/cgutman/moonlight-embedded.git
git checkout empty_buffer_fix_jessie

Then build and install using the instructions on the wiki.

Prakson, 2015-11-21:
Now 1080p 60fps bitrate 30000 works without crashes.
Thanks.

Rutherberg, 2015-11-21:
That's really nice, thanks a lot !
I'll try it tomorrow, :+1: @cgutman

PacoBarba, 2015-11-22:
I took very little time with linux and I have some difficulties installing this git . When I try to comment that it works.
Thank you very much and greetings. @cgutman

PacoBarba, 2015-11-22:
I tried to install this branch but i can´t.
cmake ../ show this message :

-- Could NOT find Freescale (missing:  FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY)
-- Could NOT find CEC (missing:  CEC_INCLUDE_DIRS) (Required is at least version "3.0")
-- Checking for module 'sdl2>=2.0.4'
--   Package 'sdl2>=2.0.4' not found
You have called ADD_LIBRARY for library moonlight-common without any source files. This typically indicates a problem with your CMakeLists.txt file
-- Configuring done
CMake Error: CMake can not determine linker language for target: moonlight-common
CMake Error: Cannot determine link language for target "moonlight-common".
-- Generating done
-- Build files have been written to: /home/osmc/moonlight-embedded/build

Prakson, 2015-11-22:
Try to install libcec-dev:
apt-get install libcec-dev

PacoBarba, 2015-11-22:
Thanks @Prakson, i use finally rbp2-libcec-dev-osmc for my osmc system and i already compile and install moonlight.
Now show this error when i launch moonlight : 

moonlight: symbol lookup error: moonlight: undefined symbol: LiInitializeStreamConfiguration

Prakson, 2015-11-22:
Try change dir to your build dir and run with
LD_LIBRARY_PATH=. ./moonlight

With LD_LIBRARY_PATH it shout use lib (moonlight-pi.so) from current dir.

Rutherberg, 2015-11-22:
I have the same problem, I installed libcec-dev, SDL2 and all the dependencies but still the missing Freescale : 

pi@raspberrypi ~/moonlight-embedded/build $ cmake ../
-- Could NOT find Freescale (missing:  FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY) 
You have called ADD_LIBRARY for library moonlight-common without any source files. This typically indicates a problem with your CMakeLists.txt file
-- Configuring done
CMake Error: CMake can not determine linker language for target: moonlight-common
CMake Error: Cannot determine link language for target "moonlight-common".
-- Generating done
-- Build files have been written to: /home/pi/moonlight-embedded/build

Same result if I add LD_LIBRARY_PATH=. ./moonlight

Prakson, 2015-11-22:
I downloaded master branch and replaced moonlight-embedded/video/pi.c with file from @cgutman branch and it works. - it is a workaround to problem with @cgutman branch.

Here are my sources:
http://78.46.195.199/moonlight-embedded.tar.gz

You can download it and compile it with these commands:
1. wget http://78.46.195.199/moonlight-embedded.tar.gz
2. tar xvf moonlight-embedded.tar.gz
3. cd moonlight-embedded
4. mkdir build
5. cmake ../
6. make -j4

After compilation in build directory will be moonlight. 
You can run it using ./moonlight

I tested it on ArchLinuxArm Rpi2.

cgutman, 2015-11-22:
I pushed a fix to my branch. You'll probably need to nuke the repo and re-checkout using the steps in https://github.com/irtimmer/moonlight-embedded/issues/198#issuecomment-158686943

Make sure you don't forget: git submodule update --init

PacoBarba, 2015-11-22:
Finally , I could reinstall OSMC have moonlight and working perfect .
I have made a clone git master branch and then I copied the file pi.c  from @cgutman branch .
Perfect, now I can put any game and and tested at key moments in which it is cut and works perfect.
To launch I need write the next line :
moonlight stream 192.168.1.XX -1080 -60fps -platform pi - bitrate 30000

Whitout expecify platform pi the graphics are horrible and slow but putting "-platform pi" works very fine.

Tanks a lot for @cgutman for your work, @Prakson for your help and at the rest.
I had to install the packages to developer to compile all, if someone needs help I can help him.

A greetings.

cgutman, 2015-11-22:
@PacoBarba can you replace src/platform.c with the file in my branch fix_platform_detection and see if you don't need "-platform pi" anymore for good performance?

Here's a direct link: https://raw.githubusercontent.com/cgutman/moonlight-embedded/07657529a906fee09dcb8509239face5d076cbf4/src/platform.c

PacoBarba, 2015-11-23:
when I get home and I try commented how it works. Thanks!

Rutherberg, 2015-11-23:
So I tried the @cgutman fix and it works, thanks :-) 

It's a great day, my first succesful compilation ! I had some trouble compiling because of missing packages and because I forgot the debian patch to compile.

I got some errors with the fakeroot debian/rules binary : 

Applying patch install.patch
patching file CMakeLists.txt
Hunk #1 FAILED at 1.
Hunk #2 succeeded at 59 (offset 6 lines).
Hunk #3 succeeded at 71 (offset 6 lines).
Hunk #4 FAILED at 74.
2 out of 4 hunks FAILED -- rejects in file CMakeLists.txt
patching file libgamestream/CMakeLists.txt
Patch install.patch does not apply (enforce with -f)
dh_quilt_patch: quilt --quiltrc /dev/null push -a || test $? = 2 returned exit code 1
debian/rules:6: recipe for target 'binary' failed
make: *** [binary] Error 1

Nevertheless, the compilation was done correctly and worked.
Sound seems to be a little bit more laggy when entering/back to Big Picture, or it is maybe a mind trick... Sound in the games seems to be fine.

I also tried to replace the platform.c in ~/moonlight-embedded/src but the result is the same, bad streaming unless you add the "-platform pi" argument.

Thanks !

PacoBarba, 2015-11-23:
Hello @cgutman. I tried to copy your platform.c and compile again and works ok.
Now don´t need add the "-platform pi" and runs the games smootly whiout crashes.
I tested in a Rpi2 with osmc August version runs with Jessie.
Thanks again.

Rutherberg, 2015-11-23:
Yes it works, I forgot to recompile... Thanks !

closed 2015-11-24 by irtimmer