moonlight-stream/moonlight-embedded #200: Can't compile OSMC Raspberry PI

ro8in, 2015-08-16:
When I do cmake ../

I get the following error

You have called ADD_LIBRARY for library moonlight-common without any source files. This typically indicates a problem with your CMakeLists.txt file
-- Configuring done
CMake Error: CMake can not determine linker language for target: moonlight-common
CMake Error: Cannot determine link language for target "moonlight-common".

I don't know how to fix this?

darkpavian, 2015-08-17:
Did you init/update the submodules?
You have to call "git submodules init" and then "git submodules update" inside the moonlight-embedded directory to initialize the moonlight-common submodule.

lhotamir, 2015-08-17:
I am experiencing the same problem. When I call:

    git submodule update

I get following error:

    fatal: reference is not a tree: 2d35d7262ffe79925114d4b5e331e52c6a436d9f
    Unable to checkout '2d35d7262ffe79925114d4b5e331e52c6a436d9f' in submodule path 'third_party/moonlight-common-c'

Ilos, 2015-08-17:
Try out: "Git reset --hard"
Then "Git submodule update --init"

closed 2015-09-12 by irtimmer

rhaleblian, 2016-08-21:
Thanks @darkpavian that worked for me. The update command appears to instead be
    git submodule update