moonlight-stream/moonlight-embedded #202: "Can't open configuration file hosts/192.168.1.xxx"

pdawg17, 2015-08-18:
Updated from version 2.0.2 to 2.1.1 and now when I start streaming I get this error. The stream does still start but even at 720p/30fps it is slow and choppy on my Windows 10 build. Prior to updating 720p/30fps worked fine...any ideas?

scoobyd00, 2015-08-18:
I dont know why thats happening, i was having the issue with very choppy video etc on 2.0.2 the upgrade to 2.1.1 under Windows 10 has made everything butterly smooth and I believe this has been the case for other users also.

Pilotzi, 2015-08-18:
Hi, i got the same "Problem", create a folder named hosts in your working dir and in this folder the file with your hosts ip address e.g. hosts/192.168.0.XXX.conf

this should hide the message because it's not really an error i think.

irtimmer, 2015-09-12:
This "problem" is fixed with commit 6621bb73ff1aaddcce781bb796bd641aba797f54

closed 2015-09-12 by irtimmer