moonlight-stream/moonlight-embedded #206: libudev0/1 dependency

Kalagon, 2015-08-19:
Would it be possible to list libudev1 as an alternative to the libudev0 dependency? OSMC (debian jessie) doesn't support libudev0 so I can't sinply install moonlight there. If incorporating libudev1  causes more problems than it fixes this is of course not an option.

TylerBre, 2015-08-22:
This is still very much an issue, and building from source is a headache on jessie.

Is jessie officially not supported on?

Ghostbird, 2015-08-30:
I “solved” this by installing libudev0 from http://ftp.nl.debian.org/debian/pool/main/u/udev/libudev0_175-7.2_armhf.deb
I haven't had the opportunity to test whether it works though, since I'm setting up the system over SSH, and I don't have a screen for the machine yet.

Diaoul, 2015-09-05:
Any progress on this? I'm interested in using the package than building from scratch myself.

baudren, 2015-10-28:
also interested in this: right now the package installation does not work on a fresh OSMC installation. An `apt-get install moonlight-embedded` greets me with:

~~~
The following packages have unmet dependencies:
 moonlight-embedded : Depends: libudev0 (>= 146) but it is not installable
E: Unable to correct problems, you have held broken packages.
~~~

Now that the PR from @dodslaser was merged, and the new release is out, I was expecting it to work out of the box. Thanks in advance for helping solving this!

dodslaser, 2015-10-30:
Hmm, maybe sources.list has changed since I installed OSMC, because i can get libudev0 via apt no problem.

I'm currently working on a Kodi add-on and wrapper for running moonlight-embedded under OSMC. Moonlight-embedded built against OSMC's libs is a possibility being considered.

In the meantime you could try installing libudev0 manually, or adding a repo that provides it to sources.list.

dodslaser, 2015-10-30:
Oh, and you probably shouldn't bother @irtimmer with OSMC specific questions. He develops against raspbian AFAIK, and OSMC is **very ** different from raspbian.

My PR only fixed the fact that it was impossible to build/install moonlight-embedded on OSMC without modifying the package. Not to make it install out of the box, or make sure it runs stable. For that we need to build and test against our own libs.

baudren, 2015-10-30:
Ok thanks @dodslaser for the update. I'll be looking forward to the Kodi add-on, then (I used the manual install of libudev0 as a temporary fix).

TylerBre, 2015-11-11:
+1 for a future kodi addon. would totally complete moonlight for me

irtimmer, 2015-11-24:
Jessie version is available in the repository

closed 2015-11-24 by irtimmer