moonlight-stream/moonlight-embedded #208: Cant use Xbox 360 controller at steam after close a game

vinanrra, 2015-08-29:
how to get error:

1º Start stream to Raspberry Pi 2 with Moonlight
2º Launch a game, i launched Metro 2033 Redux
3º Close game and try to move at steam menu

I can move without problems at start but after close game i cant move at steam menu

lhotamir, 2015-09-04:
This could be caused by a loss of focus to the Steam window. Try connecting a mouse to your Raspberry and try clicking inside the window.

I have a simillar problem when I start the stream. The first thing I have to do is clicking inside the Steam window, then I can use the controller just fine.

irtimmer, 2015-09-12:
This is a bug in Gamestream/Steam and not in Moonlight. Therefore this issue will be closed.

closed 2015-09-12 by irtimmer