moonlight-stream/moonlight-embedded #212: RTSP OPTIONS request failed: 111

Schism86, 2015-09-03:
I use version 2.1.1.
When I use the android version, everything is fine, but when i use moonlight-embedded with osmc or raspbian, i get the following error:

Starting RTSP handshake...connect() failed: 111
RTSP OPTIONS request failed: 111
failed: 111

Ilos, 2015-09-03:
Reinstall your Nvidia drivers and Geforce Experience and try again.

Schism86, 2015-09-04:
Thank you so much. It works now.

closed 2015-09-04 by irtimmer

H3llsing, 2015-09-25:
I got the same error and
> Reinstall your Nvidia drivers and Geforce Experience and try again.
didn't solve the issue.

I am on RPI2 with osmc and installed moonlight via lib, not self compiled.
PC side -> GeFe 2.5.14.5 (newest until now)

i could figure out, that the streaming only works to STEAM if i already opened Steam on PC and its the active Window.

Then the Handshakes works and he begins to Stream. i have only to click on my PI with my mouse inside of the Steamwindow and voila, gamepad support and everything is there.

Any ideas?