moonlight-stream/moonlight-embedded #213: Can't build Moonlight Embedded

vpopov-firechip, 2015-09-03:
Hello!

I'm trying to build Moonlight Embedded on my Raspberry Pi 2, with Raspbian Wheezy I got as SD card image from Raspberry Pi Foundation site:
https://www.raspberrypi.org/downloads/raspbian/
(version since 2015-05-05)

But I get many problems with missing packages, libraries versions (mostly ffmpeg and libavcodec) etc.

I am sure the "Compilation" instruction worked with some setup.

Could you please tell me which software versions I should install before building Moonlight, to reproduce working setup?

P.S. I promise to make my user experience report as a contribution, afterwards.

Ilos, 2015-09-03:
Did you read the wiki about the Compilation?
https://github.com/irtimmer/moonlight-embedded/wiki/Compilation

vpopov-firechip, 2015-09-03:
>>Did you read the wiki about the Compilation?
Hej! This is the "Compilation instruction" I referred to above.

Ilos, 2015-09-03:
yea.... 
what errors do you get exactly?

vpopov-firechip, 2015-09-03:
>what errors do you get exactly?

I did not keep exact logs because there was much tinkering. But the very first one was about wrong SDL lib version (should be >=2.0.4, and the one supplied with Rasbian was older).

Ilos, 2015-09-03:
SDL is not really necessary for moonlight to work on a Raspberry. 
There have to be other errors/missing packages that are not installed.

vpopov-firechip, 2015-09-03:
Then It could be possible I did something wrong (not following instruction). I'll redo it from start, logging my steps.

Ilos, 2015-09-03:
Consider reinstalling raspbian on your pi.
Not just moonlight...

vpopov-firechip, 2015-09-03:
>Not just moonlight...

Of course!

vpopov-firechip, 2015-09-03:
I re-did building Moonlight and succeeded. 

In general, I followed this instruction: http://www.instructables.com/id/Stream-games-to-your-Raspberry-Pi-2/?ALLSTEPS

But, with my modifications:

1) Run _sudo raspi-config_ and in  "8 Advanced options" -> "A3 Memory split" set GPU memory size to 256 MB instead of default 64, as they recommend it here:
http://embedded-software-architecture.com/?page_id=140

2) Instead of sudo apt-get update in "Step 2" I did:
_sudo apt-get update_
_sudo apt-get dist-upgrade_

3) Downloaded and installed instead of recommended .deb packages those (because recommended packages are absoleted now):
http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev-dev_1.4.3%2bdfsg-1_armhf.deb
http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev-tools_1.4.3%2bdfsg-1_armhf.deb
http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev2_1.4.3%2bdfsg-1_armhf.deb

4) Downloaded and built this version of moonlight: __moonlight-embedded-2.1.1.tar.xz__, instead of 2.0

In the end, I got Moonlight built and installed.

My plan is to re-do instruction from wiki (with another SD card) and see if it works, or there must be some correction on wiki.

closed 2015-09-12 by irtimmer