moonlight-stream/moonlight-embedded #215: Input option broken

mercier1200, 2015-09-10:
Hi,
I encountered an issue since moonlight was updated to 2.1.1. I am not able to use the –input parameter anymore. Since I use limelight (version 1.2.2), I always use the input parameter in order to use my xbox360 controller and only that. This command:

Moonlight stream –input /dev/input/event5* 192.168.1.6
is not working anymore. There is no text at screen and the program seems to wait for something.

* Event5 is my xbox controller, but every eventX not working anymore.

If I remove the input parameter:
Moonlight stream  192.168.1.6
This is working perfectly.

Anyone has encountered this issue??

Windows side specs: Windows 10 (tested on W7 too), nvidia drivers 355.82, GFE 2.5.14.5
Rpi side specs: Rpi2, Raspbian fresh and up-to-date.

7hunderbug, 2015-09-10:
Yes, I can confirm this issue with latest built from git source. 
Omitting the -input switch allows moonlight to work. Luckily, my xbox 
mapping file still works with the -mapping switch however, using more 
than one controller may be problematic as there would be no way to 
inform moonlight which mapping was for which /dev/input/event# device.

mercier1200, 2015-09-10:
Indeed, in my case, I’m using this parameter when I stream a GC/Wii game with Dolphin Emulator. Dolphin is only able to recognize my xbox controller when this is the only input connected with the stream. Removing the parameter adds every input (from client and server side) to the stream and my controller is not recognized in priority.

So this parameter is some kind important in my setup :)

irtimmer, 2015-09-12:
Input option should be fixed with commit 01ca45c7503db3e544f11fb472c4f65485fc6b00

closed 2015-09-12 by irtimmer

mercier1200, 2015-09-12:
thanks @irtimmer !