moonlight-stream/moonlight-embedded #216: Too many controllers detected by Steam.

Fireprufe15, 2015-09-11:
I have a Xbox 360 wireless controllers plugged in, and it works fine for navigating Steam and so forth.  The problem comes in when going into certain games, it just doesn't work.  I went into the controller settings thing in Steam, it detects like 8 controllers when I only have the one connected.

Now in most AAA games the controller works fine, presumably because they intelligently detect which controller is being used, but in smaller games it just doesn't happen.  The controller functions perfectly fine with these games when plugged directly into the host.

I am running RetroPie, with a script to open moonlight.  The controller also works fine in all RetroPie emulators, so it must be related to moonlight.

Marioheld, 2015-09-11:
@Fireprufe 15 How did you managed that Emulation Station recognizes the controller after being started? I tried to connect my PS3 controller via sixad but it only works when i connect the controller before ES and start ES afterwards with the command.

Fireprufe15, 2015-09-11:
@Marioheld Well IDK about PS3 controllers, but for the 360 controller there is a driver available.  I installed that and then it just worked.

Marioheld, 2015-09-11:
Thanks for the answer. Its always the same with the PS3 Controller. XD But 3 or 4 360 Controller also not that cheap so it's worth a try. I recently used recalbox which basicly does the same as RetroPie but is easier to install and use and it also has a perfect PS3 Controller support out of the box. But on the other side does it run with buildroot so it is very complicated to install software afterwards so i didn't managed it to run moonlight on recalboxOS so i thougt maybe i can get it run on RetroPie.

Fireprufe15, 2015-09-11:
@Marioheld I looked into it a bit, if you do:

    sudo update-rc.d sixad defaults

I don't know if you've done that already, but that makes the whole sixad thing run on startup.

Marioheld, 2015-09-20:
Got it working. Thanks. But it is a little tricky because sixad must connect the controller before ES finished launching.

theFisher86, 2017-01-19:
+1 I have the same issue.  I have a generic gamepad (Cyborg V3 Rumble Pad) hooked up to my Pi in RetroPie and Steam detects multiple controllers.  Even worse, in BigPicture and some games moonlight is sending double input!

Trinity92, 2017-02-10:
+1 Also having this issue. I am using a Logitech Dual Action controller connected to a Raspberry Pi 3 which acts as a Moonlight client and Steam detects 4 controllers... and even sends 4x as many inputs through Moonlight, making BigPicture mode almost unusable! It's not the controller's fault because it works perfectly fine when I run my Steam games from my PC directly without streaming. Any help or hint would be greatly appreciated.

Actually, if I disable the "Generic gamepad" configurator and instead enable the Xbox controller configurator in the settings, this multiple input issue is gone... but I am unable to do any custom button configurations from within Steam and I need this for Dark Souls f.ex. because the default control scheme doesn't work with my Logitech Controller.

Any help or hint would be greatly appreciated!

Toetje585, 2017-02-27:
Same here, it detects 4 controllers can't play some steam games because it generates dubble input.

theFisher86, 2017-02-27:
This appears to be somewhat fixed in the newest update to stream experimental. Multiple controllers still show up but they don't appear to duplicate the button presses any longer. Either way, pretty sure it's a steam issue.

Toetje585, 2017-02-27:
@theFisher86 Thanks will try to build it once I get home and test the outcome, it indeed seems to be a steam issue,

irtimmer, 2017-03-18:
Number of active controllers was incorrectly sent by Moonlight but fixed with 3407fc5b177a5cd4cdb621236e3b20a5ef42e19b

closed 2017-03-18 by irtimmer