moonlight-stream/moonlight-embedded #218: Controller Triggers not working correctly

Marioheld, 2015-09-20:
I used sixad to control moonlight via a PS3 Controller (original Sixaxis) which works perfectly with my custom map file except for the triggers (R2 and L2). When i start a stream the triggers value is set to 0 but when i press the triggers and leave them again the input stays at 127 (in the middle) instead of going back to 0. When pressing again they are correctly working up to 255 but again only goong back to 127. This causes that in racing games for example the break lights stays on and the breaks will not released completely.

SRWieZ, 2015-12-07:
Same here.

Tried to calibrate with jscal but that haven't solved the problem.
I use sixad to connect my PS3 controller https://github.com/RetroPie/sixad

I can do all tests you want to solve the problem if needed.

vdemonchy, 2016-01-17:
Hello, I have exactly the problem that @Marioheld describes, and can't figure out a solution yet, which is very annoying :(. If someone has a solution for this, please share :)

sympoz, 2016-05-11:
I have the same issue, using ps3 controller via sixad, abs_deadzone settings in mapping file does not help, only way to have analog triggers with games is set deadzone in game to 50%.

I am happy to try any suggestions

closed 2017-03-18 by irtimmer