moonlight-stream/moonlight-embedded #219: Moonlight not starting when using -input

lycake, 2015-09-26:
Hello,
I was trying to get my (wired) XBox 360 controller to work, but it's not quite working for me, yet.

I am using xbxdrv and have two USBs plugged in, my WiFi dongle and the 360 controller. There is only one input device listed in /dev/input/ (namely event0) and evtest shows it aswell and registers buttons just fine.

If I run without anything mapped, I get unmapped buttons, as expected. Mapping works and if I launch with mapping, but without specifying -input, it launches and I can see Big Picture, however, controller won't work, but I'm also not getting any unmapped buttons anymore.

Now if I want to start with -input specified, nothing happens, moonlight doesn't even start. This is what it looks like:

~/moonlight/moonlight-embedded-2.1.1/build $ ./moonlight stream 192.168.0.25 -720 -30fps -mapping xbjs.map -input /dev/input/event0
Can't open configuration file: hosts/192.168.0.25.conf

That's all, it stays like this until I abort with Ctrl+C, no errors. Maybe I did something wrong with the -input argument? Also do I have to configure something else in Steam or on the host machine?

Any help is much appreciated.
Thanks

lexa2, 2015-10-15:
Pretty much the same behavior here with Logitech F710 in DirectInput mode (could be handled both by xboxdrv userspace driver and by xpad kernel driver) and latest moonlight-embedded build available from official moonlight-embedded repo for raspbian. Tested with Raspberry Pi model B+ and with Raspberry Pi 2 using freshly installed and up to date Raspbian jessie. As soon as I specify "-input /dev/input/js0" or "-input /dev/input/event3" - moonlight seems to "silently hang" untill I press Crtl+C to terminate it.

Looking at the strace it seems that moonlight blocks on read ioctl with fd opened on specified input event device node.

P.S. Starting up moonlight without any input device node specified but having  "-mapping /usr/share/mappings/xbox360.conf" in place seems to provide correct mappings for F710 pad but results in several error messages regarding failure to map some axises (reverse_z and reverse_rz).

bertmckay79, 2015-10-22:
Same with PS3 and sixpair

Haris1977, 2015-10-23:
Same behaviour here:

I always use 

moonlight stream -mapping xbox.map 192.168.1.109

(with no inputs/events). It works for me. I havent compiled moonlight from beggining but i used this thread instead http://embedded-software-architecture.com/?page_id=140

Havent been able to use 2 different controllers at the same time though (e.g logitech f310 and xbox controller). Buttons for logitech do not work if xbox is present and vice versa

Spirit3s, 2015-10-26:
Yep, same problem, if you put a valid -input it will just block, if you put a invalid one, it will just give you a warning, and keep working

Spirit3s, 2015-10-29:
is this "Fix mapping for manual input selection " the commit to fix it ? maybe we just have to wait for a new release

Haris1977, 2015-10-29:
As said here (https://github.com/irtimmer/moonlight-embedded/issues/158) this will be fixed in new release

irtimmer, 2016-01-16:
Duplicate of #219

closed 2016-01-16 by irtimmer