moonlight-stream/moonlight-embedded #223: RTSP Describe Request failed:500

pdawg17, 2015-10-17:
I hadn't used this for awhile but when I went to use it today I got this error (had it working perfectly before). Can anyone confirm 2.1.1 is still working with up-to-date Nvidia drivers and Experience?

scoobyd00, 2015-10-17:
I've had this same problem on my windows 10 system.. 
Never got to find out wrist the error meant or how to solve it.
Did a fresh install of windows 10 and it started working again.. Hopefully it will stay like this

scoobyd00, 2015-10-27:
Did you fix the issue?

SunNoise, 2015-10-29:
I'm getting the same issue in win 8.1 with the driver 358.50 (GTX 960)
I have moonlight 2.1.1-1 i believe. I got it from the repository:
deb http://archive.itimmer.nl/raspbian/moonlight wheezy main

pdawg17, 2015-10-31:
I'm on wired so that's not the problem for me...

irtimmer, 2016-01-16:
RTSP Describe Request failed: 500 often requires a reinstall of the Geforce driver and Geforce Experience software and otherwise please provide a Wireshark dump

closed 2016-02-19 by irtimmer