moonlight-stream/moonlight-embedded #225: 5.1 audio and 4k in new geforce beta driver.

likelyeatenbyagrue, 2015-10-19:
Nvidia beta driver now supports 4k/5.1 audio. Would love 5.1 on my pi 2. 

http://www.geforce.com/whats-new/articles/geforce-experience-beta-update-introduces-new-gamestream-and-streaming-features

cgutman, 2015-10-25:
I've added support for 5.1 surround sound into common-c here: https://github.com/moonlight-stream/moonlight-common-c/commit/a13e68b0421814cb9acf93d4d4466e14700e4e39

It needs a small amount of work and testing to confirm that the channels are in the right order. Does anyone have a 5.1 config connected to their Pi?

likelyeatenbyagrue, 2015-10-25:
yes. so if i apt-get update i should be set?

cgutman, 2015-10-26:
You'll have to build from source. I've made the required changes in my experimental_surround branch.

Clone https://github.com/cgutman/moonlight-embedded.git then checkout experimental_surround.

You'll need to be on the latest GFE beta update. Pass the -surround option to Moonlight to have it stream 5.1 surround. I don't have a 5.1 system to test on. Anyone who can build and test, please report results.

likelyeatenbyagrue, 2015-10-26:
When running:
git submodule update --init

I get:
fatal: reference is not a tree: 81ecd32e833035d89538a5a5c78b4ad61ecdceb5
Unable to checkout '81ecd32e833035d89538a5a5c78b4ad61ecdceb5' in submodule path 'third_party/moonlight-common-c'

FYI: I bought a nvidia shield last week and am running on beta channel. So my pc is set to stream 5.1 with supported games (i have it fully working). In other words once i get this built I will be able to tell you with 100% certainty whether the pi is outputting sound like an official nvidia device. (for anyone wanting to stream 5.1, you don't need beta graphics card drivers, only beta geforce experience. Furthermore, nvidia's list of supported 5.1 games is rather limited at the moment so don't expect every game to work out of the box. List of games is here: https://forums.geforce.com/default/topic/885630/gamestream/supported-gamestream-games-for-4k-and-5-1-/ ).

cgutman, 2015-10-26:
Go to the third_party/moonlight-common-c directory and run: git remote set-url origin https://github.com/moonlight-stream/moonlight-common-c.git

likelyeatenbyagrue, 2015-10-26:
Ok got it compiled and running. Some interesting observations:

When running my shield and gamestreaming I get PCM audio no matter what. So the shield does the decoding and if the game supports surround i get it. 

With the pi I am getting pro logic (not PCM) no matter what...however when running -surround, all games go through all speakers. Even unsupported games. I am guessing this is no different than doing speaker fill on my receiver?

cgutman, 2015-10-26:
I suspected that we might have issues with PCM surround on the Pi. Others have had issues too. Pro Logic looks like stereo to 5.1 upmixing (which is what you heard).

Can you give details on how exactly your Pi is set up? Kernel version, distro, any audio tweaks you've done?

Can you paste the output of this command: tvservice -a

likelyeatenbyagrue, 2015-10-26:
Done absolutely no audio tweaks on the pi. It is hdmi out to my yamaha rx-v463 hdmi in. Then back out to my tv. 

uname -a:
Linux shellbeach 4.1.7-v7+ #817 SMP PREEMPT Sat Sep 19 15:32:00 BST 2015 armv7l GNU/Linux

lsb_release -a:
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie

tvservice -a: 

     PCM supported: Max channels: 8, Max samplerate: 192kHz, Max samplesize 24 bits.
     AC3 supported: Max channels: 6, Max samplerate:  48kHz, Max rate  640 kb/s.
     DTS supported: Max channels: 7, Max samplerate:  96kHz, Max rate 1536 kb/s.

cgutman, 2015-10-26:
Ok, you're pretty much set up the same way I am. The PCM max channels value seems like we should be in good shape. Mine is 2, presumably because I'm not hooked up to a surround system.

You probably didn't see my last edit in time. I put it below:

Can you also see whether either of these 2 commands play audio on all speakers:
speaker-test -Dplug:surround51 -c6 -twav
speaker-test -Dplug:sysdefault -c6 -twav

likelyeatenbyagrue, 2015-10-26:
First command : unknown PCM cards.pcm.surround51

Second command
Only hear front right and left.

cgutman, 2015-10-26:
Ok, so we're missing something here with ALSA it seems. You said you've got a Shield setup that works. Can you try Moonlight Android to confirm the surround logic is correct? Instructions here: http://forum.xda-developers.com/showpost.php?p=63454765&postcount=2906 (check the surround checkbox in settings)

I'll continue looking at the ALSA stuff.

likelyeatenbyagrue, 2015-10-26:
Installed beta moonlight on shield. Works perfect. 5.1 in supported games. 2.1 in non supported.

bdunks22, 2015-10-29:
I also tried to install and see if I could get it working on my 5.1 system but all I got was regular stereo. I'm not sure if I passed the -surround option to moonlight properly though.

cgutman, 2015-10-30:
I think the Pi may need a kernel patch to support 5.1 PCM output via ALSA.

bdunks22, 2015-10-30:
How do I pass the surround option to moonlight?

irtimmer, 2015-11-24:
Support for 5.1 is in master. Correct working depends on correct configuration of the connected surround set.

closed 2015-11-24 by irtimmer