moonlight-stream/moonlight-embedded #227: 1 Major Sound Problem when streamming Firefox/Chrome/IE

Haris1977, 2015-10-20:
Greetings. 

I have 1 major problem: (rasp2, raspbian image, moonlight embedded 2.1.1)

When i am trying to steam lets say firefox, and go into e.g this site http://dottv.gr/index.php/arxeio-ekpompes-on-demand/in-action-ekpomph-gia-ta-action-sport-arxeio-2/414-in-action-parkour , i have video but no sound. In my host pc i do have sound though. In youtube sites i also have sound both in my rasp and host pc.

Am i missing a configuration here?

Spamm00r, 2015-10-30:
Yes same issue here.

It is not limited to some sites. I only don't have sound, if I want to stream content from a Chrome browser window. If I open a media file locally for example in vlc, I get sound. Somehow the sound from Browsers does not get streamed. I thought it was an issue with my setup, apparently this is a limitation of moonlight itself. I hope there is a fix soon.

Haris1977, 2015-10-30:
I agree

Looking for a fix here:)

cgutman, 2015-10-30:
It's due to the new way NVIDIA is capturing audio. You can revert back to the old way with a registry edit: https://forums.geforce.com/default/topic/813691/gamestream/remote-desktop-audio-changed-in-gfe-2-2-2-or-shield-hub-4-1/post/4626872/#4626872

Haris1977, 2015-10-31:
Thanks man!

I don't know if i am going to have other sound related issues but for now It works :)

closed 2015-10-31 by Haris1977