moonlight-stream/moonlight-embedded #228: Change moonlight bitrate??

Haris1977, 2015-10-21:
How can i lower my client bitrate?

In android app you can "select target video bitrate". So how can i do that on my pi?.

I found this :+1: 

    moonlight streaming [options] X.X.X.X

What goes in the [options] section? While you can leave it empty and let everything run in the default settings there are a few reasons you may wish to tinker with the switches. Here are all the available command switches you can use in the options section.

    -720            Use 1280x720 resolution [default]
    -1080           Use 1920x1080 resolution
    -width          Horizontal resolution (default 1280)
    -height         Vertical resolution (default 720)
    -30fps          Use 30fps
    -60fps          Use 60fps [default]
    -bitrate        Specify the bitrate in Kbps
    -packetsize     Specify the maximum packetsize in bytes
    -app            Name of app to stream
    -nosops         Don't allow GFE to modify game settings
    -input          Use as input. Can be used multiple times
    -mapping        Use as gamepad mapping configuration file (use before -input)
    -audio          Use as ALSA audio output device (default sysdefault)
    -localaudio     Play audio locally

But what is the command for (lets say) 2 Mbps?

I now use

moonlight stream -mapping xbox.map 192.168.1.109

Thanks

Ilos, 2015-10-21:
It's right in the list.
In your case it should be for 2 Mbps:
moonlight stream 192.168.1.109 -mapping xbox.map -bitrate 2000

Haris1977, 2015-10-21:
What do i gain if i lower the bitrate to 2000-2500 Kbps? 

I am asking this because with my default settings in many games (e.g NBA 2K14, GRID 2) i have not 100% smooth streaming action (i am facing choppy animation most of the times) inside the games. 

Especially in NBA2K14 i have a little video "choppiness" when the screen is rolling. I am facing it with other games too

GFE: 2.5.11.45, wired ethernet

Ilos, 2015-10-21:
If you don't set a bitrate yourself then the bitrate is set, based on resolution and fps:
720p   & 30fps -> 5000 kbps
720p   & 60fps -> 10000 kbps
1080p & 30fps -> 10000 kpbs
1080p & 60fps -> 20000 kpbs

with higher bitrate the image quality improves, but the lag increases, and stuttering can occur.
If you lower the bitrate, then the image quality will get worse, but the stream may get smoother and the input lag also gets better.

Haris1977, 2015-10-21:
I see:) Thanks ! I will try change it and "play" a little bit.

And one more question: is there any way to map a 2nd controller to play with my friend?  I now use moonlight map name_of_controller.map to create the file on every start. But can i use 2 of them on the same time?

Haris1977, 2015-10-30:
I think the second controller issue (and generally the controller input fix) has been fixed in here https://github.com/irtimmer/moonlight-embedded/commit/01ca45c7503db3e544f11fb472c4f65485fc6b00

closed 2015-10-30 by Haris1977