moonlight-stream/moonlight-embedded #233: Loss Stats: Transaction failed: 32

dieterhatnkleinen, 2015-10-26:
I just cant get the stream working. Everytime I try to run it I get the following:
```
moonlight stream 192.168.178.25
Can't open configuration file: hosts/192.168.178.25
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Loss Stats: Transaction failed: 32
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Terminating RTSP handshake...done
Cleaning up platform...done
```

I compiled moonlight from the latest release.
Raspberry Pi 2 running OSMC

StephaneBehr, 2015-11-17:
I have the same thing on a RPI B+ with latest Raspbian Wheezy.
I will try uninstall / reinstall my nvidia drivers today, but I thought this was already solved, months ago.

I'll update this thread when I have done some more testing.

StephaneBehr, 2015-11-17:
OK, so... I reinstalled the latest NVidia Driver (which I already had installed). I made a complete "fresh install".

Then I add to re-pair Moonlight to Nvidia, and reconnect Nvidia Shield to my Google Account... And it is now working fine !

Gehilfe, 2016-01-13:
Same problem again since yesterday (2016-01-09) :(
I already tried to reinstall nvidia drivers but without success.
Running moonlight-embedded on rpi2 (osmc).
It happens exactly one minute after starting moonlight.

donfalcone, 2016-01-15:
Same problem with my config. After one minute the error happened

irtimmer, 2016-01-15:
Duplicate of #292

closed 2016-01-15 by irtimmer