moonlight-stream/moonlight-embedded #234: Missing dependancies?

Haris1977, 2015-10-30:
When i try to install dependencies i get the error:

pi@raspberrypi ~/moonlight-embedded-2.1.1 $ sudo apt-get install libopus0 libexpat1 libasound2 libudev0 libavahi-client3 libcurl3 libevdev2 rbp-userland-osmc libraspberrypi0 libssl-dev libopus-dev libasound2-dev libudev-dev libavahi-client-dev libcurl4-openssl-dev libevdev-dev libexpat1-dev cmake rbp-userland-dev-osmc libraspberrypi-dev fakeroot debhelper
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package rbp-userland-osmc is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'rbp-userland-osmc' has no installation candidate
E: Unable to locate package libevdev-dev
E: Unable to locate package rbp-userland-dev-osmc

Any thoughts?

RockyGateway, 2015-10-31:
I have a few things you can try:

After you cd moonlight-embedded, use this:

git submodule update --init

then run the rest.

If that doesn't work, use sudo with cmake ../, make, and make install after mkdir build.

I made a script that I use when I want to install Moonlight-Embedded if you want it, I've only tried it out on Jessie, though

Haris1977, 2015-10-31:
I used git submodule option. Then i tried to install dependancies and i get errors:

sudo apt-get install libopus0 libexpat1 libasound2 libudev0 libavahi-client3 libcurl3 libevdev2 libraspberrypi0 libssl-dev libopus-dev libasound2-dev libudev-dev libavahi-client-dev libcurl4-openssl-dev libevdev-dev cmake ibraspberrypi-dev build-essential
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libevdev-dev
E: Unable to locate package ibraspberrypi-dev

And in sudo cmake ../ i got this:

 sudo cmake ../
-- Could NOT find Freescale (missing:  FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY)
-- Could NOT find CEC (missing:  CEC_INCLUDE_DIRS CEC_LIBRARIES) (Required is at least version "3.0")
-- checking for module 'libevdev'
--   package 'libevdev' not found
CMake Error at /usr/local/share/cmake-3.2/Modules/FindPkgConfig.cmake:340 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/local/share/cmake-3.2/Modules/FindPkgConfig.cmake:502 (_pkg_check_modules_internal)
  CMakeLists.txt:23 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/home/pi/moonlight-embedded/build/CMakeFiles/CMakeOutput.log".

RockyGateway, 2015-10-31:
According to the Wiki you're suppose to use "git submodule update --init" after installing dependencies. I remember actually having issues getting that to work if I didn't install the dependencies first. Also you could try installing the latest cmake instead of using that older version:

https://cmake.org/files/v3.4/cmake-3.4.0-rc2.tar.gz

Haris1977, 2015-10-31:
Thanks i will try it when i get the time:)

But what about missing dependancies isuue?

I got 

E: Unable to locate package libevdev-dev
E: Unable to locate package ibraspberrypi-dev

when installing them

Haris1977, 2015-11-03:
Well. For my missing packages. I did had misspelled libraspberrypi-dev:( 

But i continue to have error when trying to download "libevdev-dev" as a standalone package. I get:

sudo apt-get install libevdev-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libevdev-dev

So any other way to install it?

closed 2015-11-10 by Haris1977