moonlight-stream/moonlight-embedded #236: moonlight.conf can't have values with spaces

Naatan, 2015-10-31:
Trying to launch a game by it's name using the config does not work if the game name has spaces in it. Wrapping it in quotes or escaping the spaces does nothing.

Ilos, 2015-10-31:
can you give an example of your execution command?
Are you sure, that the name is correct? Compare it with the name that is listed in Geforce Experience.

Naatan, 2015-10-31:
The config was generated using the --save option on arguments that were working fine for me when invoked directly on the moonlight executable.

This is my config:

```
width = 1920
height = 1080
fps = 30
packetsize = 1500
sops = false
app = The Witcher 3: Wild Hunt
```

Trying to launch it gives: `Can't find app The`

Ilos, 2015-10-31:
As you can see, it only reads the first word of the app: "The".
try out:
>app = "The Witcher 3: Wild Hunt"

Naatan, 2015-10-31:
As I said, I tried quoting and escaping, neither worked.

Ilos, 2015-10-31:
what is the error with the quotes?

Naatan, 2015-10-31:
`Can't find app "The`

Rutherberg, 2015-11-19:
I confirm, same problem with Diablo III here. Tried with the app number, underscore or minus to fill the space, but same result. The app name in GFE is Diablo III too. It shouldn't be a big problem but anyone for a workaround ?

Rutherberg, 2015-11-19:
So, dumb but efficient workaround : launch a non-steam game with Steam overlay.
To do so, in Steam -> Game -> Add a non-Steam game. And then launch the Steam stream, etc...
For Diablo 3, there is a trick to launch it from Steam : https://www.reddit.com/r/Diablo/comments/212cyj/how_to_play_diablo_3_without_bnet_launcherget/

irtimmer, 2015-11-24:
Fixed by @5dd55dcfd508ab428b4b6a4a218fdbab89e98db3

closed 2015-11-24 by irtimmer