moonlight-stream/moonlight-embedded #238: Debian patches broken on master

Naatan, 2015-10-31:
```
fakeroot debian/rules binary
dh binary --with quilt
   dh_testdir
   dh_quilt_patch
Applying patch install.patch
patching file CMakeLists.txt
Hunk #2 succeeded at 58 (offset 5 lines).
Hunk #3 succeeded at 70 (offset 5 lines).
Hunk #4 FAILED at 74.
1 out of 4 hunks FAILED -- rejects in file CMakeLists.txt
patching file libgamestream/CMakeLists.txt
Patch install.patch does not apply (enforce with -f)
dh_quilt_patch: quilt --quiltrc /dev/null push -a || test $? = 2 returned exit code 1
debian/rules:6: recipe for target 'binary' failed
make: *** [binary] Error 1
```

I also had to install `quilt` (not in the dependencies in the readme).

Naatan, 2015-10-31:
Fwiw it works using the v2.1.1 tag

TomFahey, 2015-11-13:
Yeah, I've got this error as well - did you manage to fix it?

Naatan, 2015-11-13:
Like I said, fixed by using v2.1.1, I did not bother digging any deeper.

TomFahey, 2015-11-13:
Ok, cheers

irtimmer, 2015-11-24:
Fixed with @5a064816f6096985585a1bb99ac9c3898b44d31b

closed 2015-11-24 by irtimmer