moonlight-stream/moonlight-embedded #240: Unable to locate package libevdev-dev

Haris1977, 2015-11-10:
I continue having errors when trying to download "libevdev-dev" as a standalone package. I get:

sudo apt-get install libevdev-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libevdev-dev

So any other way to install it?

burgler00, 2015-11-12:
Go to the last post in issue [#135](https://github.com/irtimmer/moonlight-embedded/issues/135) .  I just used those directions on a pi 2 and it worked without having to go through the install instructions in the wiki.

Haris1977, 2015-11-12:
So there is no need in installing dependencies or cmake? Only following this?

"Add this line to your /etc/apt/sources.list

deb http://archive.itimmer.nl/raspbian/moonlight wheezy main

Install the package using

apt-get update
apt-get install moonlight-embedded"

georgcmef, 2015-11-23:
Where can i get the libevdev-dev? Every link i see for the link is broken.

Haris1977, 2015-11-23:
Try:

wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev-dev_1.4.5%2bdfsg-1_armhf.deb
wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev-tools_1.4.5%2bdfsg-1_armhf.deb
wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev2_1.4.5%2bdfsg-1_armhf.deb
sudo dpkg -i libevdev*

If anyone knows the full ssh commands (from scratch till end) to compile and install the up to date latest beta moonlight-embedded (cmake, dependancies, git clone etc), please share with us

Thanks

georgcmef, 2015-11-24:
ok awesome thanks. dit compiles with no errors but i get this error when i try to run moonlight

moonlight: symbol lookup error: moonlight: undefined symbol: LiInitializeStreamConfiguration

irtimmer, 2015-11-24:
Compile instructions are available on the wiki https://github.com/irtimmer/moonlight-embedded/wiki/Compilation

closed 2015-11-24 by irtimmer