moonlight-stream/moonlight-embedded #241: EVIOCGRAB failed with error 16

Makkaan, 2015-11-14:
Hello!

I've been trying the last few days to install OSMC with Moonlight.

I get the popup to pair the machines but i get this error after that:
Can't open configuration file: hosts/192.168.1.13.conf
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...RTSP message too long
RTSP DESCRIBE request failed: -1
failed: -1
EVIOCGRAB failed with error 16
EVIOCGRAB failed with error 16
EVIOCGRAB failed with error 16

TallyTalman, 2015-11-21:
I had the exact same problem concerning the RTSP handshake failing.

Looks like this is a Nvidia related issue.

I found this program: http://www.guru3d.com/files-details/display-driver-uninstaller-download.html
It uninstalled everything Nvidia related properly in Safe Mode. For your own convenience I advise you download the driver beforehand from the official Nvidia site, so once have uninstalled everything and you get back from safe mode, all you have to do is run the installer.

This fixed the same problem immediately for me (I was messing around way to long, because I didn't realize my previous attempt to reinstall Nvidia was insufficient).

Cannot help you with your problem regarding the old PC, because most of this stuff goes way over my head, but I hope this helps you get past the RTSP handshake on your main machine, like it did for me. Good luck.

cgutman, 2015-11-24:
This is a duplicate of #245

irtimmer, 2015-11-24:
Fixed with @48f88f6d2fb2fe4a4bbd800cbbd8332e09b73a0d

closed 2015-11-24 by irtimmer

diegombeltran, 2015-12-01:
Still not working, using last OSMC packages and Moonlight-Embedded 2.1.2 from Jessie repo