moonlight-stream/moonlight-embedded #243: arch linux + Orange pi -  Platform 'default' not found

k-zy, 2015-11-15:
hello,

I'm trying to use moonlight-embedded on arch linux 
"Linux OrangePi 3.4.39 #1 SMP PREEMPT Mon Oct 12 12:02:29 CEST 2015 armv7l GNU/Linux"

I have follow this : https://github.com/irtimmer/moonlight-embedded/wiki/Packages

When i want to pair my computer, i have this error : 
[root@OrangePi ~]# moonlight pair 192.168.1.123
Platform 'default' not found

k-zy, 2015-11-15:
My bad, this is not a raspberry Pi platform.
Maybe there is a way to port moolight-embedded on Orange Pi (or Banana pi) ?
I'm using Loboris's  linux distribution form :
 http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=342&extra=page%3D1

k-zy, 2015-11-18:
I can't succeed to compile moonlight embedded for my platform.
Does the linux Arch ARM package is design only for Rpi and I.mX device ?

irtimmer, 2016-04-03:
Duplicate of #232

closed 2016-04-03 by irtimmer

phyrexia, 2016-07-08:
OrangePi is different than Raspberry , i think you could add support for this boards.