moonlight-stream/moonlight-embedded #244: Randomly becomes unpaired and can't re-pair

BrainSlugs83, 2015-11-18:
So, I set everything up yesterday and it was all working fine -- today I'm failing to connect at the RTSP stage, it just hangs.

Figured, I'd try re-pairing the device, went into GFE->Preferences->SHIELD and clicked "Forget devices".

On the RPi2, typed in "moonlight pair [hostname]" -- and it says the usual "Please enter the following PIN on the target PC: 8335" -- followed **IMMEDIATELY** by "Successfully paired".

On the PC there was never any pop-up, etc., and it's not actually paired -- all commands -- including "LIST" fail as well, saying "You must pair with the PC first".

It's clearly caching something, and that sucks, because I can't re-pair it to my PC now.

snakecc, 2015-11-26:
make sure only one user account is active on pc.

NeoArmageddon, 2015-12-04:
I have the same issue. I also made sure I have only one user account active.

7hunderbug, 2015-12-04:
Have any of you tried deleting the certificate files that are created by moonlight-embedded when pairing is completed and which can be found in the directory you run moonlight from? As well as forgetting devices in GFE, you probably need to delete the certificate files too.

NeoArmageddon, 2015-12-04:
Found the problem in the meantime and was not moonlight related.

GFE was just bugged as usual. Reinstalling GFE and copying all needed Shadowplay and Streaming .dlls from the installer to the installation directory solved the problem.

closed 2016-04-03 by irtimmer

theFisher86, 2016-04-04:
@NeoArmageddon could you explain exactly what you did when you Re-Installed GFE and which version you're using **Right Now**?  I've been having a terrible time getting this working and I have a feeling it's all GFE's fault as well but all I've done was uninstall and re-install version 2.10.2.40 b/c everything post 2.10 hasn't worked.