moonlight-stream/moonlight-embedded #245: RTSP handshake failed... RTSP message too long (describe failed: -1)

TallyTalman, 2015-11-21:
I am officially giving up on solving this myself given my lack of know-how and insufficient resources (none) on this issue online. Let me explain the problem:

Yesterday I starting installing Moonlight on my Rpi2 (which also has retropie/kodi installed on top of Raspbian Jessie). I tried compiling at first and all went seemingly well, but once the moment of truth arrived I got the following error: 

Initializing platform...done
Resolving host name...done
Starting RTSP handshake...RTSP message too long
RTSP DESCRIBE request failed: -1
failed: -1

Big Picture mode starts on my PC, but no magic on the Pi.

TallyTalman, 2015-11-21:
Never mind... This is why I should just keep trying...

Had a feeling it wasn't Moonlight, so did some more digging and figured it had to be Nvidia related (most problems with moonlight are apparently). 
Found a much better uninstaller on guru3d (http://www.guru3d.com/files-details/display-driver-uninstaller-download.html) which did a proper uninstall in Safe Mode, reinstalled all the Nvidia stuff and it worked right away!

Hours of messing around and 'right after' posting an issue I find the solution.. Should've had more faith in the terrific people contributing to this great project!

Naph, 2015-11-21:
Currently having the same issue after using the driver removal tools you have supplied. Was there any other steps you took?

Sendoushi, 2015-11-21:
So... Uninstall through guru3d.com uninstaller, reinstalled through GeForce Experience and it is still not working.

```
Initializing platform... done
Resolving host name... done
Starting RTSP handshake... RTSP message too long
RTSP DESCRIBE request failed: -1
failed: -1
```

cgutman, 2015-11-21:
So this error is actually Moonlight's fault. It completely slipped my mind when I saw Nvidia had shipped GFE 2.7 to production. There's an update required in moonlight-common-c for compatibility with 2.7 that increases the RTSP message limit that is being hit: https://github.com/moonlight-stream/moonlight-common-c/commit/b1f861ccfdde1391f18becb67c64836658800a0b

The other Moonlight clients had versions released weeks ago with the fix already included back when GFE 2.7 was in beta, so it never crossed my mind that moonlight-embedded might have an issue.

@irtimmer seems like time to pull the latest commits in moonlight-common-c and make a new release. GFE 2.7 is in production via the forced auto-update mechanism since Thursday, so users with the old release can't stream once GFE updates itself.

TallyTalman, 2015-11-21:
Just tried it again and got the same error again, so my enthusiasm was a bit premature last night. Apologies for that. 
But great to hear the issue is known and fixable. In the mean time I'll get an older GFE version and continue messing around, figuring stuff out, which is part of the fun tbh.

For those interested: I uninstalled Nvidia again using before mentioned uninstaller and downloaded driver 358.50 released 10/7/15 (which comes with GFE 2.5.14.5) and it seems to work.

Haris1977, 2015-11-21:
I am using moonlight-embedded with **GFE 2.4.5.57** and **358.50** driver and i confirm it works as expected. I have also disabled (through the etc/hosts file) the auto-update function which makes gfe auto updates when a new release is out.

As TallyTalman said it should also work with **GFE 2.5.14.5** but I think that there is a "conflict" of moonlight-embedded with the 2.5.15.54 version (i couldnt pair my pc)

So a new update of moonlight-embedded should come out:)

substring, 2015-11-24:
@cgutman @irtimmer I couldn't stress more the need of a new release ...

cgutman, 2015-11-24:
@irtimmer Debian, Arch, and source packages are broken for any new user downloading GFE or a new driver from Nvidia's website. GFE's auto update will break people's existing installations too, the next time they open GFE. We've already had 3 bugs filed on this same issue.

I think we've gotten all the big bugs fixed IMHO. Master is working great for me and others. I think it's time for a release.

irtimmer, 2015-11-24:
Duplicate #241

closed 2015-11-24 by irtimmer