moonlight-stream/moonlight-embedded #248: Steam Big Picture > Games crash...

Sendoushi, 2015-11-22:
Hi,

Streaming the Steam Big Picture Mode works well. I play "most" of the games well but there are others that crash.

So, from my library, the games that crash:
```
Borderlands 2    # Crashes after starting game and initial story
Spelunky            # Crashes on enter
```

H3llsing, 2015-12-13:
i have the same problem with this game.
Already tried to start via Steam, without Steam (direct call in gameslist with moonlight).
starting on the pc without streaming works well. i also tried to connect one gamepad to the pc but still crashes when streaming.

`Magicka      # After starting`

havnar, 2016-02-08:
I have the same problem with most games that I've tried so far.
Can get into big picture mode but there it dies.

cgutman, 2016-02-08:
There's a bug that will need to be addressed with a GFE update.

https://forums.geforce.com/default/topic/909598/gamestream/gamestream-broken-with-steam-and-big-picture-mode/

irtimmer, 2016-02-09:
Duplicate of #292

closed 2016-02-09 by irtimmer