moonlight-stream/moonlight-embedded #249: Support for Banana Pi/Pro (CedarX library)?

jul13579, 2015-11-23:
Dear irtimmer,

I just wanted to ask, if there is any possibility to run moonlight-embedded on a Banana Pi/Pro, which uses the CedarX video library? I would really like to use Moonlight on my Banana Pro!

Greetings,
jul13579

Jhauck1, 2015-12-05:
Agreed !!

JokinenK, 2016-01-16:
If the support for Banana Pi is implemented at some point, the same code should also work with little to none extra work with Orange Pi boards. As far as I know they use the same VPU and GPU as the Banana Pi (CedarX & Mali400).

irtimmer, 2017-06-11:
Moonlight embedded has added support for VDPAU. So hopefully someone can check it is working with https://github.com/linux-sunxi/libvdpau-sunxi (VDPAU backend for Allwinner/CedarX) so this issue can be closed.

closed 2017-06-18 by irtimmer