moonlight-stream/moonlight-embedded #252: Not such a "smooth" experience. Any thoughts?

Haris1977, 2015-11-26:
I don't know if this is a directly moonlight/network/drivers related issue, but since I cannot localize the problem itself, I just wanna try to get some input here.

My Setup:
PC: Intel core 2 quad q6600 @2.4 Ghz, 4gb RAM, GTX 750 Ti, gfe 2.4.5.57, driver 358.50, 1920x1080, 60hz
Client: RPI 1, oc to high, Raspbian jessie, speedlink torid wireless gamepad
Network: Router is connected through ethernet hub to rp1 (wired)

While everything works fine (even managed to play with 2 controllers) almost all games are not as "smooth" (in terms of fps experience) as i expected. E.g when i play PES 2015 while in my pc the game is smooth, there is a ***fps drop*** (from e.g 60 to 30 or less) ***for about 4-5 seconds***. Then this probl dissapears for some seconds. Then it appears again. It is like a sudden fps drop. I can see that to almost any game.

I am using the -30fps command 

moonlight stream -30fps -mapping xbox.map 192.168.1.109

The problem remains when: 

1) I lower bitrate from default to 3000
2) I run moonlight to 720p 60 Hz (haven't tried 1080)

Any thoughts appreciated

Haris1977, 2015-11-27:
Maybe this http://www.2shared.com/photo/QaN0uCDe/bug.html has smth to do with my problem?

So maybe a video-audio sync problem?

Haris1977, 2015-11-29:
I tested both iperf (in idle state, 88 mbit) and nethog (when playing dirt2 game). Nethog showed me this

http://www.2shared.com/photo/yIsF1jOC/nethogs.html

Are these values normal?

cgutman, 2015-12-02:
Can you paste a log from moonlight?

Haris1977, 2015-12-02:
How could i do that?

cgutman, 2015-12-02:
Run moonlight and pipe the output to a file like:

moonlight stream &> log.txt

Haris1977, 2015-12-02:
You mean a command smth like this?

moonlight stream -30fps -mapping xbox.map 192.168.1.109 &> log.txt

??

cgutman, 2015-12-02:
Yeah, then after you see stuttering for a bit, exit with Ctrl+Alt+Shift+Q. Then put the log on pastebin or something.

Haris1977, 2015-12-03:
Ok, this is my log file

http://pastebin.com/PiQTiXEw

(the first line fail is because i haven't attached my usb gamepad)

rstanislav, 2015-12-03:
I dont think its same issue as i have - there is no any errors about rtp packet drop or decode in your log file.
Also on RPi there is hardware decoder that used by this project, in my case it uses software ffmpeg + SDL to display it.

In your log file there is errors about ALSA (audio) output and that can be the reason.
To test it i can suggest for example to use -localaudio option with moonlight - audio will be played on host pc and not on rpi.

Also - how do you connect audio to rpi ? Via USB ? If yes then there is maybe problem - from my tests long time ago there is still unresolved issues, atleast on first rpi model - RPi randomly drops USB packets and that causes loss of sound(artifacts while playing), breaks video from webcams etc... i have tested on 2 RPi 1 - and it was the same on both. Currently i can only say that this issue no longer present on rpi 2 b+.

Haris1977, 2015-12-03:
Well i am having the same "issue" in both my rpi1 b model and rpi2 b model. I am using hdmi as the output of my pi's (as most of people). Haven't tested local audio in my pi though. I will test and see if this is the problem. But if this is true, it is not convinient not to have audio from your pi:(

rstanislav, 2015-12-03:
rpi 2 b+ 100% dont have issues with usb, anyway by disabling audio you can check if its causing fps drop or not.

Haris1977, 2015-12-03:
I will test it,thanks:)

Haris1977, 2015-12-03:
-localaudio doesn't work for me however i use it as an argument (i always get sound)

Haris1977, 2015-12-04:
Maybe this one?

https://forums.geforce.com/default/topic/894952/geforce-drivers/nvidia-streamer-service-causing-20-cpu-usage-/post/4722605/#4722605

cgutman, 2015-12-05:
You're streaming with the -30fps flag. This means you're absolutely going to see judder when it's sampling from a 50 FPS source.

Haris1977, 2015-12-05:
I removed the -30 fps flag and now streaming is much better. It is not 100% smooth (i get some jitter when i turn left/right in dirt3) but better. What do i need for 100% smooth experience?

cgutman, 2015-12-05:
It's going to stutter if your frame rate drops below 60 on your PC. The further it goes from 60, the more stuttering you'll see. Try lowering settings on your PC to ensure your frame rate isn't dropping as much.

Haris1977, 2015-12-06:
While i have a constant 59-60fps in my host pc, i continue to have some jittering which is vissible especially in racing games. I must admit it is much better now (i removed the -30fps argument) but not perfect.

An other problem is that i get **randomly jittering** without any reason (e.g in pes2009, but i am sure it happens in other games too). Is this a moonlight-embedded problem or drivers/network related?

Haris1977, 2015-12-12:
Since there is no feedback about my problem i contacted nvidia support team. Their answer (copy paste)

**I understand from your email that you are experiencing issues with streaming games from your computer to your device using Moonlight program. I apologize for the inconvenience caused and I'll do my best to help with the issue.

Since the Frame Rate loss does not occur on the Host PC, the issue could be related to either the Moonlight app itself or the Wireless network connectivity. The program may be built up on the GeForce GameStream concept, but the program is not developed by NVIDIA. Hence, only the program developers can assist with the issue pertaining to this setup or the program.

Best regards, 
Ayub,
NVIDIA Customer Care**

So..maybe an app related issue? Since there is no feedback from the devs here i think this app is dead..

cgutman, 2015-12-12:
Wow, I'm surprised they even gave any help at all. I should frame that and hang it on my wall ;)

Moonlight is not their problem to deal with. I try to discourage Moonlight users from bugging Nvidia, because I don't believe it will ever lead to good things. Even if it is an issue with GFE, they aren't going to give it much credit coming from a Moonlight user, since they don't have fully knowledge of the software end-to-end.

I'm not sure why you think the app is dead, given that there was a release just 3 weeks ago and commits 8 days ago. It's still very much active.

I think the reason you're not getting responses is that nobody knows what the problem is. It works fine for many other people so there must be something unique in your setup.

In software development, sometimes you just have these (as of yet unexplained) gremlins. It might take days to figure out what the cause is, and that's a tough ask from someone doing this in their spare time.

Anyway, back on topic. I'd really like to see you get a solid 60 FPS on your PC and see if the issue still aries. Play in 720p resolution on your PC on low settings if you have to.

This is why:
There's a sampling issue in play similar to Vsync stutter. GFE is sampling the frame buffer every 16 ms. If your game is operating a 50 FPS, that's a frame every 20 ms. That means roughly every in-game 5 frames (20 / (20 - 16) = 5 frames), GFE samples the same frame twice (for 33 ms on screen). This happens once every 100 ms (5 * 20 ms / frame = 100 ms). You see that as video stutter. 

If you have a solid 60 FPS on the PC and still see stuttering on the Pi, then we've got something interesting.

Haris1977, 2015-12-12:
Thanks @cgutman for your feedback again. That is exactly my problem as you dedcribed. While i get a solid gameplay i get stuttering on the pi (the truth is that i have a sensitive eye in terms of stuttering) but it is trully there. Lets focus on dirt3 for example. Here are my in game settings:

http://www.2shared.com/photo/rdzfEHXw/20151212_113715.html

Perfectly smooth on my pc (60-61 fps, tested with many fps counter apps even with steam fps counter), not good in pi. It is like a resampling issue or smth.

I am here for any help

Kheiryk, 2015-12-13:
Hi,
i start a game without fps limit and i see the fps in GFE.
always 60 fps with Rpi 2 in ethernet connection.

Haris1977, 2015-12-13:
So..you are having 60 fps **BOTH** in your host pc **AND** your rpi??

Haris1977, 2015-12-14:
@cgutman i ve downloaded your moonlight android app from google play store (https://play.google.com/store/apps/details?id=com.limelight&hl=el). I ve tested it with my galaxy s3 phone. This app **runs much better** without any (noticable) fps drops (well, there are some minor glitches but not so many). I ve tested it in 720p, 60fps, 4 mbit bitrate, wifi 2.4G wireless modem (TP-LINK TD-W8901G) if this helps..

So i am sure it is smth on my pi side (app/pi settings)

Haris1977, 2015-12-28:
I dont know if this helps, but a lot of nvidia's shield tv owners are facing same jittering problems 

https://forums.geforce.com/default/topic/885856/gamestream/dropped-frames-ethernet-and-shield-tv-/1/

slowshutterfilms, 2016-01-12:
I'm having this exact problem, but I can't even stay in Big Picture for more than a couple minutes without it crashing. I constantly get ALSA errors. I read in an earlier thread that installing an older bootloader fixes the issue, but I don't know how to do that and the specific one they suggest is not available.

Haris1977, 2016-01-13:
So do y think that this video microstuttering is alsa error dependant??

slowshutterfilms, 2016-01-13:
That's what I'm thinking. Somehow the audio latency is lagging out the whole connection.

Haris1977, 2016-01-13:
Well..this was my initial thinking but unless @irtimmer implements this https://github.com/irtimmer/moonlight-embedded/issues/271 we wont be able to know what is going on..

P.s i have abbandoned moonlight right now as i am waiting for steam link device hopping (?) to be better..

diagonali, 2016-01-29:
I'm surprised that there isn't (or I haven't seen) anyone address the ALSA errors. I don't know if they're acknowledged ( @irtimmer ?) to be a Moonlight problem or else a Pi problem (haven't seen any mention myself) but it would make sense if there was some sort of connection with that and the stuttering a few of us (me included) are getting on the Pi.

@slowshutterfilms  Also saw the suggestion to downgrade the bootloader but unless I have step by step commands to type into Putty I'm not going to be able to try it unfortunately. I did give it a go (I'm running OSMC) but whatever I did it bricked my installation. I can only hope that either OSMC or Moonlight will get an update that will resolve the strange bugs causing less than expected performance - the performance that some using the Pi report getting great and fairly smooth results.

Haris1977, 2016-01-29:
Just a thought: i am no away from this project as i am the owner of a steam link right now. So, i am not aware of any progress about this project. But this might help nvidia's users:

I had similar problems with my steam link. Micro-jitterings every few seconds (y could see it especially in game replays like dirt3), even if i lowered my game settings to 800x600 with low details. 
So that made me think that this may not be the client's problem. Digging it a little further, i managed to discover that my problem was my gpu's **capture method**. I changed it to NVFBC (google for more info about nvfbc vs nvifr) and now my games are pretty smooth!

In order to see what capture mode your gpu is into see this thread:

https://steamcommunity.com/groups/homestream/discussions/0/490125103626704042/#p5

(download the zippyshare file and run the check_status.bat)

This is for nvidia users of course..

rstanislav, 2016-01-29:
In my case i just switched to using steam in home streaming.. no more of this shiity lags, sluttering and other problems - it uses hardware cpu\gpu encdoing and one of most important things - it has support of keyboard and mouse, and video delay way lower.

As of this project - i simply dont get it - author spend  A LOT of time, develop a open source project and then don't support it ? What the point ? 1 simple bug with sluttering makes it unusable. This is just plain stupid imo.

7hunderbug, 2016-01-29:
@rstanislav WOW! Rude much?!! What if the problem isn't a bug with moonlight? Had you even considered that possibilty before expressing your odious opinion? Do you really think that if @irtimmer knew what the solution was he would withhold it from you? What if the problem actually lay with NVidia's implementation of their streaming protocol, their drivers or with your hardware (which is what I suspect to be true as I and many others don't have the problem that you're whinging about).

In future I suggest that if you want a free and open-source software project to succeed, and you want help from the community, many of them contributing in their own free time, you'd be better off being more considerate and producing proper bug reports, detailed logs and detailed evidence of your "bug".

I'm glad you got what you wanted using an alternative method (albeit without using a Raspberry Pi) - this project will be much better off without you and you childish and *STUPID* comments like the one you've just made.

slowshutterfilms, 2016-01-29:
@rstanislav How did you use in-home streaming on the raspberry pi?

slowshutterfilms, 2016-01-30:
Okay, I found a workaround for the time being in issue #292 . Add steam as it's own app in GFE, then launch it from moonlight.

rstanislav, 2016-01-30:
@7hunderbug it goes the same for moonlight - authors spend even more time and because of few simple bugs it dont works as it should.

There is alot of people with same problem, and i dont belive that you have it working as it should - 60 fps at 1080p without sluttering, image tearing, frame loss and delay.

You can't support project if it's author dont want to accept that help. 
Look at the date of first post - its over 2 months ago and there is no any movement ..

@slowshutterfilms i'm using small pc as streaming target, not rpi.

irtimmer, 2016-01-31:
@rstanislav people don't write open source software for others but for themselves. With the benefit that others can reuse and improve the code. Currently it works perfectly in my own setup (rpi1 archlinux 1080p 30fps, because 60fps is too heavy for the rpi1) so I find it very useful. Also it works far better then Steam (with nvfbc) on my Linux laptop.

As far for this bug there is currently just not enough debug information and most of the causes arise because of problems with GFE or the firmware provided by Raspbian.

xfechx, 2016-09-07:
I am having the same issue, both on an Nvidia Shield Tablet (connected to a TV via HDMI) and also through a Samsung Galaxy s7 Edge. 

There is a microstutter on every game, regardless if the game is running at 60fps. 

When I run the game directly from my graphics card to HDMI, all run butter smooth. 

My specs:
ASUS MINI NVIDIA GTX 970 running through thunderbolt via AKITIO enclosure ( eGPU )
Macbook Pro running Windows 10 with a GHz Intel Core i7 processor (2.5 ~ 3.5 ghz )
SAMSUNG SSD drive connected via USB 3

I am really hoping that gamestream / moonlight will be able to run butter smooth without micro-stuttering. I am assuming that this is a gamestream problem, can anyone recommend me the best gamestream version to use?

closed 2017-03-18 by irtimmer

fabioneves, 2018-11-26:
Opening this old thread.. since it's happening to me, I've tried parsec at the same time, I've added FPS on the screen for my ps vita and PC and there's no FPS dropdowns at all.
In parsec the experience is 100% smooth. I've uploaded a video of what happens, it's really a small stutter / jerkyness.

Happens at 7sec, 24sec, that micro-stutter / jerkyness.

[image]