moonlight-stream/moonlight-embedded #254: Dropping video frames, image sluttering\lagging

rstanislav, 2015-11-27:
Main pc is win7, 4790k OC'ed, dual SLI gtx 970 also OC'ed, 16gb ram.
On pc fallout 4 runs on all maxed settings at 2560x1600 butter smooth, no lags no matter how complex scene is..explosions etc...

But when streaming, i get frame loss = sometimes image even freeze for 1-2 seconds and i'm not doing anything at all... just running around, at same time if i look at main streaming pc - its butter smooth there...

PC i'm streaming to - debian, intel NUC D54250WYK - core i5 with 16 gb ram, top shows about 45-50% cpu load, network is gigabit, no losses, tryed different switches, even direct connection to streaming pc....
With low bitrate like 7000 its better, but not that much...at 30000 its a litle bit worse..

what i'm doing wrong ?

Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Decode failed
Decode failed
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped end of a frame
Discarding RTP packet after queue overgrowth
Network dropped end of a frame
Discarding RTP packet after queue overgrowth
Network dropped end of a frame
Discarding RTP packet after queue overgrowth
Network dropped end of a frame
Discarding RTP packet after queue overgrowth
Network dropped end of a frame
Discarding RTP packet after queue overgrowth
Network dropped end of a frame
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Decode failed
Decode failed
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Decode failed
Decode failed
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Discarding RTP packet queued for too long
Network dropped beginning of a frame
Discarding RTP packet after queue overgrowth
Network dropped beginning of a frame
Invalidate reference frame request sent
Decode failed
Decode failed
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Decode failed
Decode failed
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Discarding RTP packet after queue overgrowth
Network dropped beginning of a frame
Invalidate reference frame request sent
Decode failed
Decode failed
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Decode failed
Decode failed
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Decode failed
Decode failed
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Decode failed
Decode failed
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Discarding RTP packet after queue overgrowth
Network dropped beginning of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Discarding RTP packet after queue overgrowth
Network dropped beginning of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Invalidate reference frame request sent
Loss Stats: Transaction failed: 32
./ml: line 2:  2642 Segmentation fault      /root/moonlight-embedded/build/moonlight stream -1080 -60fps -app "Fallout 4" -fullscreen -surround -bitrate 20000 192.168.1.37

rstanislav, 2015-12-02:
ADD: Tryed today moonlight-pc (jar precompilled) - and its like ten times worse - complety unplayble, huge frame loss. Also tryed to use it on iMac, but on imac i can't connect xbox 360 receiver and can't figure out how to use mouse and keyboard - official shield from nvidia has support for it, moonlight - does not it seems, futuremore - while streaming session is active keyboard and mouse on host PC is also "disconnected" - is there any way to make it work ? (atleast on host pc)

Haris1977, 2015-12-02:
I have also frame drops, and while in my pc the game runs pretty smooth (with lower details of course as my pc is obslete), in my rpi there is a definite fps loss every now and then and especially in modern games - like dirt3, pes15 etc (it is like when you see a ntsc fps 29.97 movie which stutters every now and then). It is not my network and i think it is not my pi either. To check, i used **android moonlight app** in my phone, which is pretty good without any drops (except for my non 5Ghz router)

Have opened a same issue here:

https://github.com/irtimmer/moonlight-embedded/issues/252, if someone can look at it..

rstanislav, 2015-12-02:
I think i'm 1 step closer to solution - in file
moonlight-common-c/limelight-common/RtpReorderQueue.h
if i modify
#define RTPQ_DEFAULT_MAX_SIZE   16
#define RTPQ_DEFAULT_QUEUE_TIME 40

to something like

#define RTPQ_DEFAULT_MAX_SIZE   3000
#define RTPQ_DEFAULT_QUEUE_TIME 3000

Its way more better...5 min of play and no errors\frame freeze or frame loss...at high bitrate like 30000 there is sometimes strange encode artifacts tho... but no image sluttering\micro freezing.

default values way too low, at 30 mbps encode rate its = about 3.6 megabytes per second and if each packet is 1500 bytes in size (its can't be larger by default) it means that there is 2400 packets per second, buffer of 16 is just simple stupid...
If i incease buffer without increasing max queue time its also dropping packets (because sometimes it waits for full frame or host pc encoder can lag etc...).
Now it way way better, but needs more testing time.. will try to do it in next days.

Haris1977, 2015-12-02:
@rstanislav I don't know if we are facing the same kind of problem but if you find a solution please let @irtimmer know about it:)

cgutman, 2015-12-02:
@rstanislav That queue doesn't contain all packets sent from the server. It holds a reserve of packets while waiting for delayed or out of order packets to be delivered. As long as packets are arriving in order, nothing is queued. Increasing the limits means that moonlight won't notify the PC of a dropped frame until much later, causing much worse stuttering when packet loss does occur. That's why you see the increased artifacting.

The issue seems to be that ffmpeg isn't fast enough to decode the stream in real time on your hardware. When that happens, the receive buffer fills up and begins dropping packets. Because the SDL decoder is set as direct submit, the decoding and rendering happens on the receive thread. IMHO, irtimmer is breaking the API contract wrt direct submit decoders with SDL - they're supposed to be fast and this one isn't. I don't think it should have CAPABILITY_DIRECT_SUBMIT set. Remove that flag in src/video/sdl.c and see if that improves things.

rstanislav, 2015-12-02:
Didn't have much time to check.. its night in moscow :) but its perfect now! I was thinking about sdl (was searching for problem and found few forks of this project with gstreamer implementations - thinking about trying it) - will try to do more tests tomorrow and will write results. Thank ALOT for answer/help **cgutman**

rstanislav, 2015-12-03:
Found the reason - ffmpeg uses only single thread...and even core i5 can't decode 1080p60 fast on single core... if its possible to make it use all 4 cores then it will be ok.

Also i do not understand why SDL is used... ffmpeg can display decoded frames itself..

Haris1977, 2015-12-03:
@rstanislav  do y think we are facing the same kind of problem? Have opened a same issue here:

#252

cgutman, 2015-12-03:
@rstanislav The last two parameters to ffmpeg_init() are swapped. Try swapping those around and it should improve things a bunch.

rstanislav, 2015-12-03:
Installed ubuntu 15.10 instead of debian, got all intel drivers build tools libs etc... (5 hours wasted lol)

Tryed to compile Hansa064 fork with gstreamer (3 hours wasted, with 2 of them searching why alsa dont work)
... got black screen during stream... but have audio.

Now compiled last x254 and ffmpeg, changed params in sdl.c (also changed thread param from 2 to 4) - and now its working way more better, i can see sometimes small fps drop, but its huge defference... and from top i can see that moonlight uses about 120-130% cpu and 2 cores loaded by ~ 50% and other 2 by ~ 10-20%. But still its not perfect - i get

Discarding RTP packet after queue overgrowth
Network dropped middle of a frame

sometimes... about 15 messages during 5 min play.

Such a huge time wasted with such small result... but atleast now its playable...
Next things to do - i will try to fix gstreamer and then implement hardware VAAPI decode...

ffmpeg is a dead end imo(+SDL...)... both VAAPI and VDPAU does not support anything except direct paint via opengl... its possible to get frame decoded via hardware back from GPU, but it req. to transfer it via memory and then paint etc.. and at 1080p60 that single task(transfer and paint via software) can kill core i5 performance complety...

It should be painted directly via VAAPI/VDPAU api directly via GPU core on screen, like its doing it in imx6\RPi etc....

rstanislav, 2015-12-03:
Tested last presets for about 30 min - its way better and playble, but there is some bug's.
First - sdl buffer should be increased atleast to ~ 120*1024 or it will cause segfault with "Video decode buffer too small" error, second and very important:
**cgutman** - i think, input buffers in RTP queue should be increase to atleast ~ 800 packets(at 30 mbs its about ~ 350ms buffer based on my calculations above)  and about 160-260ms max time - currently in software decode ffmpeg is causing problems, but nvidia SHIELD is supposed to work over internet, where udp packets easy arrive out of order with 100+ ms time, and it will cause unplayble slutter\frame loss with current 16/40 preset in moonlight-core code.

irtimmer, 2016-01-01:
Duplicate of #252

closed 2016-01-01 by irtimmer